qmk/qmk_configurator #52: Provide means to download .bin file

mechmerlin, 2018-04-09:
On keyboards such as the whitefox, flashing is done by doing
```
dfu-util -D ../../.build/whitefox_default.bin  -S mk20dx256vlh7
```

The QMK Configurator currently has no means to download the required .bin file.

yanfali, 2018-04-09:
Ditto for clueboard 60.

yanfali, 2018-05-11:
Looking at the code, this is actually built into the backend, so the front end is ready mostly for when the backend can do this. We may want to change the name of the download button based on metadata we know about the board.

yanfali, 2018-06-03:
Talked this over with @skullydazed and it looks like we can already handle this via the API. I'm going to file a new enhancement bug to rename the download hex button to download firmware.

closed 2018-06-03 by yanfali

holtenc, 2018-06-23:
I see this change has been implemented, but for the TADA68 it still downloads a HEX instead of a BIN

yanfali, 2018-06-23:
Nope. Sorry. Waiting in all changes still.

jibreil, 2018-06-26:
@yanfali any idea if/when this functionality will be implemented, myself and others are looking for an easy way to get the `.bin` for a TADA68 without manually building QMK via the command line.

yanfali, 2018-06-26:
@jibreil hopefully in the near future. @skullydazed is a bit busy right now, but hopefully he can get it over the line once he's freed up. It's mostly a backend issue as the front end doesn't have control over what bits are sent over the API.

jibreil, 2018-06-26:
@yanfali 👍