qmk/qmk_configurator #66: Numpad in Keycode Selection is Missing 3

mechmerlin, 2018-04-20:
[image]

It's missing 3!!!

Thank you to isabella on the r/mk discord for reporting.

yanfali, 2018-04-20:
haha, it was actually correct before from the code perspective, but it's always said 4. 4 is unlucky in asian countries.

```diff
-      { name: '1', code: 'KC_P1' },
-      { name: '2', code: 'KC_P2' },
-      { name: '4', code: 'KC_P3' },
+      { name: '1', code: 'KC_P1', keys: 'num_1' },
+      { name: '2', code: 'KC_P2', keys: 'num_2' },
+      { name: '4', code: 'KC_P3', keys: 'num_4' },
```

mechmerlin, 2018-04-26:
Can confirm.

closed 2018-04-26 by mechmerlin