qmk/qmk_configurator #72: Looks terrific, fails to compile for Input Club K-Type Keyboard

dmgm, 2018-04-26:
I created a base layer only, by clicking a key, then pressing the physical key on the keyboard.
I set the fn key to me MO1. This was done on macOS with chrome as the browser.

I exported the resulting json file (which I can provide if needed).

I pressed compile. Which failed with the output. pasted below.

> # Input Club K-Type
> 
> Firmware for custom keyboard PCB with TKL Key Layout
> 
> Keyboard Maintainer: [Kaleb Elwert](https://github.com/belak)  
> Hardware Supported: Input Club K-Type   
> Hardware Availability: [Input Club](https://input.club/k-type/), [Massdrop](https://www.massdrop.com/buy/massdrop-x-input-club-k-type-mechanical-keyboard?utm_source=linkshare&referer=WJJG5M)
> 
> Make example for this keyboard (after setting up your build environment):
> 
>     make k_type:default
> 
> See [build environment setup](https://docs.qmk.fm/build_environment_setup.html) then the [make instructions](https://docs.qmk.fm/make_instructions.html) for more information.
> 
> * Sorry there is no default for the k_type keyboard... yet!
> * Sending k_type:K-type-base-layer with KEYMAP
> * Received job_id: 6e241c26-734d-4a85-8564-e5a0f45eaabb
> * Running . . . . . . . . . . . . . .
> * Finished:
> QMK Firmware 0.6.18
> 
>  Some git sub-modules are out of date or modified, please consider runnning:
>  make git-submodule
>  You can ignore this warning if you are not compiling any ChibiOS keyboards,
>  or if you have modified the ChibiOS libraries yourself. 
> 
> 
> 
> tmk_core/chibios.mk:38: lib/chibios-contrib/os/common/startup/ARMCMx/compilers/GCC/mk/startup_k20x7.mk: No such file or directory
> tmk_core/chibios.mk:40: lib/chibios/os/hal/hal.mk: No such file or directory
> tmk_core/chibios.mk:46: lib/chibios-contrib/os/hal/ports/KINETIS/K20x/platform.mk: No such file or directory
> tmk_core/chibios.mk:79: lib/chibios/os/hal/osal/rt/osal.mk: No such file or directory
> tmk_core/chibios.mk:81: lib/chibios/os/rt/rt.mk: No such file or directory
> tmk_core/chibios.mk:87: lib/chibios/os/common/ports/ARMCMx/compilers/GCC/mk/port_v7m.mk: No such file or directory
> tmk_core/chibios.mk:89: lib/chibios/os/hal/lib/streams/streams.mk: No such file or directory
> makek'.  Stop.
> Makefile:534: recipe for target 'k_type:K-type-base-layer' failed
> make: *** [k_type:K-type-base-layer] Error 1
> ake finished with errors
> 
> 
>

JackCuthbert, 2018-04-26:
Also failing with Whitefox Truefox

dmgm, 2018-04-30:
I looked at the configurator today and couldn't find the k-type listed.  Has it been removed?
If so, is that the solution to this issue?

mechmerlin, 2018-04-30:
ARM boards require a different compile process and also a different file to be downloaded. We removed it for now until we get that support in.

dmgm, 2018-04-30:
OK. I'll close the issue, and open a "Please add support for ARM keyboards like K-Type and Whitefox"

closed 2018-04-30 by dmgm