qmk/qmk_configurator #73: MO(X) doesn't get exported/imported properly

mechmerlin, 2018-04-26:
When creating a keymap with Fn key. Placing anything in MO() results in it being set to 0 when exported and imported.

yanfali, 2018-04-26:
I just tested this, and it looked ok.

```
{"keyboard":"1up60rgb","keymap":"mine","layout":"LAYOUT_60_ansi_split_bs_rshift","layers":[["KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","MO(8)","KC_NO"],[],[],[],[],[],[],[],["KC_TRNS"]]}
```

yanfali, 2018-04-26:
Can you document the steps to reproduce?

mechmerlin, 2018-04-26:
1. Make a ok60 keymap (yes the layout isn't done yet). 
2. Make MO(1) with an appropriately populated layer 1 to jump to. 
3. Export Keymap
4. Open another tab and navigate to config.qmk.fm. Select the correct board. 
5. Import said Keymap

MO(1) should now be MO(0).

yanfali, 2018-05-11:
OK, I think this was the focus bug.

yanfali, 2018-05-13:
One way to solve this may be to force a focus change before we export/compile the keymap to trigger the event. I'm not sure how reliable this would be. The other possible way to deal with this is to actively poll the field while it is being edited.

yanfali, 2018-06-03:
I think I know how to solve this issue, but I would prefer to defer the fix to when we move to a self a more flexible UI toolkit than jQuery. The work around for now is to ensure you have left the input box so the event is captured.

buzzlighter1, 2018-09-23:
I have the same bug. Imported layout > changed MO(0) to MO(1) > exported, and it didn't change it.

yanfali, 2019-02-24:
This should be working now.

closed 2019-02-24 by yanfali