qmk/qmk_configurator #87: Support for ARM keyboards like K-Type and Whitefox

dmgm, 2018-04-30:
Please add support for the Input Club keyboards, for example K-Type and Whitefox

yanfali, 2018-06-03:
@skullydazed is working on adding ARM support back to the API, as soon as have this the API will begin advertising ARM based boards again.

dmgm, 2018-06-03:
Could we have an intermediate step? I have successfully manually flashed QMK to the K-Type. Could we make it so the configurator allows you to save a configuration, which would then be manually flashed?

yanfali, 2018-06-03:
Before it was disabled, you could sort have done this with the download source button. I think that would be too confusing for novice users though, which is why it was temporarily disabled.

yanfali, 2018-06-03:
@skullydazed is there any way to get a cli tool which takes the keymap JSON and converts it to source like the API?