qmk/qmk_configurator #91: Tooltips for Controls

mechmerlin, 2018-05-04:
People are getting confused with the controls, especially the import/export button, thinking it is for hex files. We need to add tooltips to our controls briefly describing what they do and what they are for.

yanfali, 2018-05-04:
maybe Import JSON and Export JSON? Just like Download Hex?

mechmerlin, 2018-05-04:
I'll take this one, sounds within my skill level

yanfali, 2018-05-04:
sorry, I already did it

yanfali, 2018-05-04:
I'll hold off. You do it and add the accept bit from my branch
https://github.com/yanfali/qmk_configurator/commit/859e94e28c780bd16210f8aca342aca46cf6b88b

yanfali, 2018-05-04:
The accept attribute prevents people from loading file types that aren't JSON.

yanfali, 2018-05-04:
OK, we discussed this a bit on discord and the general consensus for now is to add the accept attribute to block hex files being loaded, and to use #92 to prevent people import JSON files that are from kbfirmware.

Longer term, QMK needs to register a mime type with IANA and then being using it consistently across the configurator/toolbox.

yanfali, 2018-05-07:
fixed.

closed 2018-05-07 by jackhumbert