qmk/qmk_configurator #98: qmk compile failing

karni24, 2018-05-07:
trying to compile layout for hexfile but it keeps failing, have tried playing with loayout numerous times to see if that was the issue and it fails everytime

[plzwork (1).txt](https://github.com/qmk/qmk_configurator/files/1981712/plzwork.1.txt)

yanfali, 2018-05-07:
thank you! We'll take a look and see what we can figure out.

yanfali, 2018-05-07:
Hi, we may have found the culprit. Follow up question. Have you ever hand edited the JSON file? Thanks

karni24, 2018-05-08:
no, only thing ive done to the file is rename it to .txt

yanfali, 2018-05-08:
That's good. A few codes appeared in the file that aren't expected by the
API. We haven't quite figured out how they got in there. If you want to if
them. Replace the [, ], . codes in the file with KC_LBRC, KC_RBRC and
KC_STOP. It's not clear how the symbols got into the save file, so that's
definitely a bug. After I fixed those things the file compiled. Hope that
helps.

yanfali, 2018-05-08:
@karni24 good news. I've figured out how to duplicate your bug. If you configure an ANY key, that lets you type in any text into a key. This is an expert mode feature, so we don't check what you type into there. This will be saved faithfully into the keymap.json and then when you import it, because of a different feature I added for keyboard input, it imported the key incorrectly and even displayed the symbol.

To prevent this happening in the future, We're adding a check for the length of the keycode, and we'll ignore ones that are shorter than our expected length. Thanks for reporting this bug!

yanfali, 2018-05-11:
I believe this issue is now fixed. @karni24 let us know if you still have this issue, but if not please close the issue. Thanks

closed 2018-05-11 by karni24