qmk/qmk_configurator #107: Downloading hex file ends up with a blank text file

jibreil, 2018-05-11:
I'm trying to flash my GH60 and have done so multiple times in the past few days by compiling and downloading the hex from qmk_configurator.

Today when I press `Download .hex`, it tries to download `download.txt` which is a blank text file with nothing in it and with a 0KB file size. The `.json` i was trying to use can be seen [here](https://u.teknik.io/Flsjd.json) and the result in the console after pressing `Compile` can be seen [here](https://p.teknik.io/Simple/WNHDM) and [here](https://u.teknik.io/6vt60.PNG).

yanfali, 2018-05-11:
Sorry @jibreil my bad. It should be fixed in the next PR.

jibreil, 2018-05-12:
Thanks @yanfali, seems to be fixed now 👍

closed 2018-05-12 by jibreil