qmk/qmk_configurator #121: Fails to compile when using LCtl_T and RSFT_T

Arrowmaster, 2018-05-18:
I created a keymap for Atreus62 using LCtl_T and RSFT_T. It fails to compile when using these but succeeds after removing them.

{"keyboard":"atreus62","keymap":"mine","layout":"LAYOUT","layers":[["KC_GRV","KC_1","KC_2","KC_3","KC_4","KC_5","KC_6","KC_7","KC_8","KC_9","KC_0","KC_MINS","KC_TAB","KC_Q","KC_W","KC_E","KC_R","KC_T","KC_Y","KC_U","KC_I","KC_O","KC_P","KC_BSPC","LCtl_T(KC_ESC)","KC_A","KC_S","KC_D","KC_F","KC_G","KC_H","KC_J","KC_K","KC_L","KC_SCLN","KC_QUOT","KC_LSFT","KC_Z","KC_X","KC_C","KC_V","KC_B","KC_N","KC_M","KC_COMM","KC_DOT","KC_SLSH","RSFT_T(KC_ENT)","KC_LCTL","KC_LGUI","KC_LALT","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO"]]}

yanfali, 2018-05-18:
Thanks for the report. We have a fix incoming. Good catch.

yanfali, 2018-05-18:
Should be fixed now. Please try again. Thanks

yanfali, 2018-05-19:
@Arrowmaster please feel free to close the issue once you've verified it's working. Thanks

Arrowmaster, 2018-05-19:
Importing the old json file still failed to compile. Removing the bad keys and recreating them in the configurator corrected the issues and it now compiles. Did not flash hex to keyboard and test functionality.

closed 2018-05-19 by Arrowmaster