qmk/qmk_configurator #129: Ergodox Layout broken in Configurator

drashna, 2018-05-25:
Not only does it generate an error in the API error list .... it doesn't show right in the configurator.... despite the fact that the info.json is correct (tested it with comebody's previewer first, actually)

You can view it here.  The top row of each half has one too many keys.  This affects ALL of the layouts, actually. 
https://config.qmk.fm/#/ergodox_ez/LAYOUT_ergodox

However, this is not reflected in the info.json nor the ergodox_ez.h file:
https://github.com/qmk/qmk_firmware/blob/afacd42368e0dc7627a695508f15598b38429c63/keyboards/ergodox_ez/ergodox_ez.h#L109-L254

https://github.com/qmk/qmk_firmware/blob/afacd42368e0dc7627a695508f15598b38429c63/keyboards/ergodox_ez/info.json

@yanfali suspects that it may be a parser issue with the macros. 
@skullydazed, if you wanted to take a look at this bizarre issue, when you have time?

Cybolic, 2018-06-08:
The issue seems to be that comment blocks are read as key definitions.

E.g.

```
#define KEYMAP_PRETTY_80(                                                                              \
    /* left hand, spatial positions */     /* right hand, spatial positions */                         \
    L00,L01, ...
```

Outputs:
```
LAYOUT_ergodox_pretty_80: {
  key_count: 82,
  layout: [
    {x: 0, w: 1, label: "/*lefthand", y: 0},
    {x: 1, w: 1, label: "spatialpositions*//*righthand", y: 0},
    {x: 2, w: 1, label: "spatialpositions*/\nL00", y: 0},
    {x: 3, w: 1, label: "L01", y: 0}, ...
```

yanfali, 2018-06-08:
That looks like a parser issue. @skullydazed looks like we need to strip comments out :(

drashna, 2018-06-08:
Also, this PR will fix the issue: https://github.com/qmk/qmk_firmware/pull/3052
I'd merge, but it's my PR...

spytecinc, 2018-09-07:
Hey guys! Just wanted to report that the same issue is occurring for all 3 layouts for Frosty Flake, and I'm guessing others as well.

Since this issue is still open, I thought I'd mention it here. @drashna I can try to mirror the changes you made for the Frosty Flake layout and submit a PR if you think that's still a good idea?

drashna, 2018-11-04:
This was fixed in the qmk repo, so it's no longer an issue with that board.  

Not closing, because it should be improved to prevent this issue in the future.

yanfali, 2018-11-04:
@skullydazed this would be a perfect time to try transferring an issue between repos.

closed 2020-10-25 by drashna