qmk/qmk_configurator #150: How to

aidenhx, 2018-06-16:


aidenhx, 2018-06-16:
Hey, guys! I can't find an option for configuring a macro. Can anybody help me? Thanks!

mechmerlin, 2018-06-16:
Macros are currently not supported in the Configurator. The only way right now is to download the source and code it up the old way.

yanfali, 2018-06-16:
Hi @aidenhx, sadly this is true for now. However, you can use the configurator to generate your initial keymap and then compile it. Then hit download source to get a starter keymap.c which you can then use as a base for your macro programming. Hope that helps. We fully intend to add macros but we're not there yet.

tastethesoup, 2019-08-08:
Any recent updates? Would love this functionality in the configurator

fireworm71, 2019-11-20:
Would adding "strings only" be a good first step?

yanfali, 2019-11-20:
> Would adding "strings only" be a good first step?

It's getting closer. QMK CLI project was started by @skullydazed and this paves the way to having API support for things like macros.

kmanwar89, 2020-05-02:
Several months later, any updates? This function would be great.

If not, can anyone guide me on where in the keymap.c I would create the macro? Is there a specific syntax it expects?