qmk/qmk_configurator #153: look at why planck doesn't compile

yanfali, 2018-06-17:


noroadsleft, 2018-06-18:
Culprit is the `EXT_PLV` keycode. https://github.com/qmk/qmk_configurator/blob/ee789f0a895d62df4384ae15ee673a56d0c4e751/keymaps/planck_default.json#L11

Compiles without errors if you delete the keycode.

----

Defined in as a macro in qmk_firmware.

- https://github.com/qmk/qmk_firmware/blob/44660e77a4e4c3f6ab180df1f8135ba0691dc012/keyboards/planck/keymaps/default/keymap.c#L38
- https://github.com/qmk/qmk_firmware/blob/44660e77a4e4c3f6ab180df1f8135ba0691dc012/keyboards/planck/keymaps/default/keymap.c#L152
- https://github.com/qmk/qmk_firmware/blob/44660e77a4e4c3f6ab180df1f8135ba0691dc012/keyboards/planck/keymaps/default/keymap.c#L238

yanfali, 2018-06-18:
@jackhumbert what do you want to do? If we leave this out the keymap will compile in configurator.

yanfali, 2018-06-18:
@noroadsleft you want to open a PR with the code removed and let's get @jackhumbert to review it?

noroadsleft, 2018-06-18:
I'm willing to do so, but then the default in Configurator won't match the default in qmk_firmware, which I'm not keen on.

I have no idea how the Plover support works, so can't help there.

yanfali, 2018-06-18:
Yeah, I think it's unavoidable though for the time being unless we want a
non working config.

yanfali, 2018-06-18:
Awesome work @noroadsleft

closed 2018-06-18 by yanfali

noroadsleft, 2018-06-19:
Thank you, @yanfali.