qmk/qmk_configurator #157: default keymap name of "mine" cannot be easily deleted

mechmerlin, 2018-06-18:
Using Chrome on macOS 10.13.4

When trying to delete "mine" by backspacing over one character at a time, starting from e, down to m. As soon as m is deleted, "mine" is shown again. The only way to properly delete it is to highlight "mine" and then delete. 

Coincidentally, this behavior is not consistent. If you have successfully made a keymap name such as 
"no_more_iso", highlighting it and deleting will generate "mine" again. 

This is quite annoying.

yanfali, 2018-06-18:
Yeah, I'm aware of this behavior. I can look into changing it. The current work around is to select all and then start typing. it's to do with the reactive model checking the value. I think I can change it to allow empty and only substitute in mine if you leave it blank.

yanfali, 2018-06-20:
OK, closing this as fixed.

closed 2018-06-20 by yanfali