qmk/qmk_configurator #170: Stuck on "Sending" step

gwerbin, 2018-07-02:
I submitted [this layout](https://gist.github.com/gwerbin/d7f0f0047259d64a0b3660a55c663d95) to the [QMK configurator](https://config.qmk.fm), but it seems to be stuck on the first stage,

```
Loaded 4 layers and 332 keycodes. Defined 0 Any key keycodes
KBD75
===

Keyboard Maintainer: QMK Community  
Hardware Supported: KBD75 PCB rev 1 and 2 
Hardware Availability: https://kbdfans.myshopify.com/products/kbd75-keyboard-set?variant=35638534029

The KBD75 has two revisions. Revision 2 has a USB C port and allows a numpad layout. `.hex` files made
for KBD75 revision 1, will still work for revision 2. `.hex` files made for revision 2 (excluding the numpad)
will also still work on revision 1. Both revisions share the same switch matrix but have switch holes in 
different areas. 

Make example for this keyboard (after setting up your build environment):

    make kbd75/rev1:default
    make kbd75/rev2:default

See [build environment setup](https://docs.qmk.fm/build_environment_setup.html) then the [make instructions](https://docs.qmk.fm/make_instructions.html) for more information.
* Sending kbd75/rev1:kbd75/rev1_layout_ansi_mine with LAYOUT_ansi
```

Any thoughts? Is something wrong with my layout, and this is causing some kind of error scenario? Is the backend API down?

yanfali, 2018-07-02:
Looks like a bug on the import. I'll take a look in a bit and get back to you. I think I know what the problem is.

yanfali, 2018-07-02:
OK, there's a bug on export.  To work around it for now, please edit the keymap name and remove 'kbd75/' and I'll work on fixing this. Thank you for reporting this!

yanfali, 2018-07-02:
@gwerbin a fix is in place, please try again in about 5 minutes. Thanks.

closed 2018-07-02 by yanfali

gwerbin, 2018-07-02:
Tested and it works. That was fast, thank you!