qmk/qmk_configurator #175: Create a folder exclusion list for keyboard selection drop-down

noroadsleft, 2018-07-12:
Some "keyboards" able to be selected in the Configurator are not actually valid keyboards, but rather directories that contain the files for keyboards. The containing directories should be excluded from the drop-down menu.

@yanfali @mechmerlin

mechmerlin, 2018-07-12:
Thanks for filing this, here's my list of what we need to blacklist

atom47, christmas_tree, crkbd, deltasplit75, e6v2, eagle_viper, eco, ergo42, ergodash, ergotravel, fortitude60, fourier, hadron, helix, iris, jc65, kbd75, kinesis, launchpad, lets_split, levinson, mechmini, meira, minidox, nyquist, octagon, orthodox, planck, preonic, redox, s60_x, vitamins_included, zen

noroadsleft, 2018-07-12:
Okay, here's my proposed exclude list:

- atom47
- chibios_test
  - This a valid keyboard in a sense, but not sure it serves a purpose being in Configurator.
- crkbd
- deltasplit
- e6v2
- eagle_viper
- eco
- ergo42
- ergodash
- ergotravel
- fourtitude60
- fourier
- hadron
- helix
- iris
- jc65
- kbd75 (rev2 needs better layout support IMO)
- kinesis
- launchpad
- lets_split
- lfkeyboards
- mechmini
- meira
- minidox
- nyquist
- octagon
- orthodox
- planck
- preonic
- ps2avrGB
  - This folder should maybe be deleted from qmk_firmware, or perhaps kept/moved somewhere as a guide/reference for porting ps2avrGB-based boards. All the boards it says it supports are elsewhere in qmk_firmware.
- redox
- s60_x
- vitamins_included
- viterbi
- zen

noroadsleft, 2018-07-12:
Keyboards on my list that are not on @mechmerlin's:

- chibios_test
- lfkeyboards
- ps2avrGB
- viterbi

Keyboards on @mechmerlin's list that aren't on mine:

- levinson

The Levinson, I meant to add but forgot to note.

ChibiOS_test and ps2avrGB I noted in my previous comment.

IMO LFKeyboards should be deleted because it's a vendor folder rather than a keyboard. Looking at qmk_firmware, I'm not actually sure why it's showing up in the drop-down.

yanfali, 2018-10-01:
merged.

closed 2018-10-01 by yanfali