qmk/qmk_configurator #183: How to configure adjust layer?

aidenhx, 2018-08-03:


yanfali, 2018-08-03:
Hi @aidenhx this is one of the few keyboards where we're have difficulty supporting the current keyboard code directly. Adjust/raise/lower are implemented in code rather than as keycodes, so you would have to use a combination of  layers to simulate something similar.

We're working on it, but it's not ready yet.

In the meantime, please join our discord so you can ask questions in real-time and hopefully we can get you where you want to go.

peterkos, 2025-01-12:
It looks like this has been done in QMK (and the adjust mode seems to be implicit if the `LOWER` and `RAISE` are pressed simultaneously) -- https://docs.qmk.fm/features/tri_layer#keycodes