qmk/qmk_configurator #184: Help Overlay

mechmerlin, 2018-08-07:
**Some common problems observed on the configurator** 

1. People forget to choose an appropriate layout, resulting in confusion when filling up extra "spaces" in the keycap. 

2. After keymap creation, people are unsure of where to go next. Some think that the resulting .json from `Export Keymap` is the final result. 

3. Extension of point #2, users don't understand what the word compile means and proceed to press Download Firmware which can't be clicked. 

**Request**

It would be helpful to have some kind of Help Overlay which guides users to the different parts of the configurator that needs user input. 

Step 1: Choose Keyboard
Step 2: Choose Layout
Step 3: Make Keymap
Step 4: (Optional) Export Keymap
Step 5: Compile
Step 6: Download Firmware

yanfali, 2018-08-07:
https://github.com/pulsardev/vue-tour looks promising

zekth, 2019-06-30:
Having a Tour mode seems cool tho. Vue-tour has no real updates since last year, but seeing the issues it seems ok. I can try to tackle it.