qmk/qmk_configurator #185: [TADA68 ISO] Fails to generate working binary

DarkDefender, 2018-08-17:
Hi, I tried out the configurator and generated the following keymap for my tada68:
[conf.zip](https://github.com/qmk/qmk_configurator/files/2297163/conf.zip)

At first it didn't want to compile (failed at the "Running" step), so I exported the keymap and loaded it again. After that, it seems like the compile goes through without any errors and I get a 4kb .bin file.

However it doesn't work at all when I flash my tada68 with it. I don't know what the problem could be as I do not see any error output from the configurator. If I click "dowload source" i get:
```
{"message":"Compile job not finished or other error."}
```

yanfali, 2018-08-17:
Hi @DarkDefender the tada68 support is still incomplete unfortunately. @skullydazed is working on it and I hope to have a solution for you presently. I'll update this issue once we've figured it out.

DarkDefender, 2018-08-17:
Ah, I see. No problem, I took and compiled my own `.bin` file from source. So I'm in no hurry really.

yanfali, 2018-09-18:
@DarkDefender can you give it another go please? We just added .bin support. You will be our first user.

DarkDefender, 2018-09-18:
Oh, I see. I think I will have some time tomorrow when I can test it out. Thanks for the heads up

DarkDefender, 2018-09-19:
Using the same json file as before, it now generates working binaries. However there is one huge flaw. The middle row is shifted to the left. So "asdf" -> "sdfg".

I don't know where to find the generated `.c` file. However this is my working config file that I used when I generate the firmware myself: https://gist.github.com/DarkDefender/d4274023bec1e02e8618d384c53340d7

yanfali, 2018-09-19:
Try download source. It'll contain the keymap.

DarkDefender, 2018-09-19:
Oh right, I did download the source before. But I didn't expect it to create a new layout in the ordinary place somehow...

So here is the generated source (edited for formatting):
https://gist.github.com/DarkDefender/eee9cfc8e57059bb804fbd9ee143e788

If we compare it to my working config, we can see that the backslash/iso hash key is in the wrong place. While it would be logical to have it in it's actual physical location, it seems like it should be in the same place as the ansi backslash key (which is somewhat logical as it will act as the ansi key if you switch to a ansi keyboard keymap).

It also explains the shift to the left.

yanfali, 2018-09-19:
Thanks for looking into this. It sounds like the layout is wrong in the main repo. I'll chat with the firmware folks and try and get this fixed up.

yanfali, 2018-09-19:
OK. Looks like this is a firmware issue. Going to close this now as fixed.

https://github.com/qmk/qmk_firmware/pull/3943 to follow thread on there @DarkDefender

yanfali, 2018-09-19:
@fcoury also tested the bin support on an ANSI Tada and it worked fine. Thanks @fcoury

closed 2018-09-19 by yanfali

fcoury, 2018-09-19:
Thank you all for the amazing and continued work @yanfali