qmk/qmk_configurator #190: Configurator error log: more detailed error messages.

noroadsleft, 2018-08-31:
The current error log for the Configurator has a lot of these:

```json
{
    "severity": "error",
    "message": "QMK Configurator Support Broken:\n\nNo layouts defined."
},
```

Doesn't say which keyboard is broken, which makes troubleshooting nigh impossible.

Additionally, there's ones like this:

```json
{
    "severity": "error",
    "message": "Error: clueboard/66_hotswap/gen1: LAYOUT: Number of elements in info.json does not match! info.json:70 != LAYOUT:73"
},
```

which are more helpful, but need debugging on their own, because the message is unclear about which number came from which source. Is the `70` from the `info.json` file, or the matrix macro? I can't ever remember, so I end up counting them manually.

I propose something more like:

```json
{
    "severity": "error",
    "message": "Error: clueboard/66_hotswap/gen1: LAYOUT: Number of elements in info.json does not match! info.json/LAYOUT contains 70 elements, but gen1.h/LAYOUT contains 73."
},
```

yanfali, 2018-08-31:
@noroadsleft we also have this https://yanfali.github.io/qmk_error_page/

noroadsleft, 2018-09-02:
@yanfali I look at this sometimes, but from what I've seen, a number of these only fail for lack of API support, rather than actually being broken.

Tap Dance in particular breaks a number of these. Compiles perfectly if run from downloaded QMK source, but fails in Configurator.

skullydazed, 2018-09-17:
I yanked the configurator errors from the log in this commit: https://github.com/qmk/qmk_api/commit/9d106a9780b90dc9d277f47102f4bd5cf55e7e9a

I've opened an issue on yanfali/qmk_error_page to switch that to a richer API, so we can still access that information.

skullydazed, 2018-09-17:
The error log changes are out, is it looking better?

On the number of elements does not match message, my intent was that the numbers convey how many elements each side has. In that case info.json has 70 entries, while LAYOUT requires 73. Do you have any suggestions for how I can make that more clear?

yanfali, 2018-09-18:
OK, closing this now the feature has been moved to https://github.com/yanfali/qmk_error_page

closed 2018-09-18 by yanfali

yanfali, 2018-09-18:
@noroadsleft that's an API issue, I think @skullydazed needs to clear out or regenerate the data over time. The UI doesn't cache anything.