qmk/qmk_configurator #195: Chibios compiling and downloading

yiancar, 2018-09-26:
As I understand it the issue to get Chibios based boards is twofold.

Firstly, the backend does not support pulling the necessary git modules. For example Chibios, Chibios-Contrib etc. I imagine there should be a way for periodically to check those repos and pull when needed.

Furthermore I have been informed that for some reason the frontend cannot let you download .bin files (@yanfali has some suggestions on this).

I would consider this relatively critical to the support of STM32 based boards as most end users don't want to manually compile there firmware.

yiancar, 2018-09-26:
I did not know that the API was not here:)

closed 2018-09-26 by yiancar

yanfali, 2018-09-26:
No problem