qmk/qmk_configurator #197: loading default layers

yanfali, 2018-09-30:
https://api.qmk.fm/v1/keyboards/nyquist

returns a datastructure with a field called keymaps

```json
"keymaps":["yshrsmz","DivergeJM","losinggeneration","333fred","kim-kim","danielhklein","pitty","jojiichan","default"]
```

assuming a keymaps field exists, search for default and use it to load the keymap. If no default exists attempt to load the first one in the list.

https://api.qmk.fm/v1/keyboards/<keyboardpath>/keymaps/<keymap name>

Returns a data structure that looks like this:

```json
{
   "layout_macro" : "LAYOUT",
   "layers" : [
      [ "KC_1" ],
      [ "KC_F1" ],
      [ "KC_SPC" ],
   ],
   "keymap_folder" : "qmk_firmware/keyboards/nyquist/keymaps",
   "keyboard_name" : "nyquist",
   "keymap_name" : "default"
}
```

Use the layers info to pre-populate the keymap on configurator when there's a matching LAYOUT macro.

skullydazed, 2018-09-30:
```
"keymaps": {
     'LAYOUT': ["yshrsmz","DivergeJM","losinggeneration","333fred","kim-kim","danielhklein","pitty","jojiichan","default"],
    'LAYOUT_60_ansi': ['layout1', 'layout2']
}
```

yanfali, 2018-09-30:
Yeah this makes more sense as we have multiple layouts. I like it.