qmk/qmk_configurator #201: Build firmware for Ergodox Infinity

douglas, 2018-10-09:
Hello!

I was trying to build a firmware for a Ergodox Infinity but it didn't work, here's the build log:

```
# ErgoDox Infinity

The Infinity is two completely independent keyboards, and needs to be flashed
for the left and right halves seperately.  To flash them:

  - Make sure you are in the top-level qmk_firmware directory

  - Build the firmware with `make ergodox_infinity:keymapname`

  - Plug in the left hand keyboard only.

  - Press the program button (back of keyboard, above thumb pad).

  - Install the firmware with `sudo make ergodox_infinity:keymapname:dfu-util`

  - Build right hand firmware with `make ergodox_infinity:keymapname MASTER=right`

  - Plug in the right hand keyboard only.

  - Press the program button (back of keyboard, above thumb pad).

  - Install the firmware with `sudo make ergodox_infinity:keymapname:dfu-util MASTER=right`

More information on the Infinity firmware is available in the [TMK/chibios for
Input Club Infinity Ergodox](https://github.com/fredizzimo/infinity_ergodox/blob/master/README.md)

## Infinity Master/Two Halves

The Infinity is two completely independent keyboards, that can connect together.
You have a few options in how you flash the firmware:

- Flash the left half, rebuild the firmware with &quot;MASTER=right&quot; and then flash
  the right half.  This allows you to plug in either half directly to the
  computer and is what the above instructions do.

- Flash the left half, then flash the same firmware on the right.  This only
  works when the left half is plugged directly to the computer and the keymap
  is mirrored.  It saves the small extra step of rebuilding with
  &quot;MASTER=right&quot;.

- The same as the previous one but with &quot;MASTER=right&quot; when you build the
  firmware, then flash the same firmware to both halves.  You just have to
  directly connect the right half to the computer.

- For minor changes such as changing only the keymap without having updated
  any part of the firmware code itself, you can program only the MASTER half.

- It is safest to program both halves though.
* Sending ergodox_infinity:douglas with LAYOUT_ergodox
* Received job_id: cda06e0d-7bcb-4746-8168-60617b72ef9d
* Running . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . .
* Finished:
QMK Firmware 0.6.137
WARNING:
 Some git sub-modules are out of date or modified, please consider running:
 make git-submodule
 You can ignore this warning if you are not compiling any ChibiOS keyboards,
 or if you have modified the ChibiOS libraries yourself. 

Making ergodox_infinity with keymap douglas

quantum/visualizer/visualizer.mk:77: lib/ugfx/gfx.mk: No such file or directory
makek'.  Stop.
Makefile:529: recipe for target 'ergodox_infinity:douglas' failed
make: *** [ergodox_infinity:douglas] Error 1
Make finished with errors
```

If I'm not mistaken, what I had to do to build it locally was to run `make git-submodule` to fetch the submodules.

yanfali, 2018-10-09:
@douglas sorry, but there's an issue with the ugfx library. I think the license prevents us from distributing it. Only end users are allowed to download it and compile against it. I'll talk to the other folks on QMK to possibly remove boards that use this library from the configurator because of this limitation.

yanfali, 2018-10-09:
@skullydazed @jackhumbert do you know what the story is with ugfx? is this like arm and we just need to git-submodule the library, or is there an license issue?

douglas, 2018-10-09:
Hmmmmmmm, is there a way for `configurator` export a `keymap.c` file then? It would help qmk configuration greatly! Thanks for you hard work! `Configurator` helps me a lot ❤️

douglas, 2018-10-09:
Also, this is weird as I don't have this library installed here - are you sure it isn't an outdated configuration/checkout on the server?

drashna, 2018-10-09:
Well, the configurator has a "download source" which is everything you need to set up a build environment. 

As for the lib, it's a git submodule, specifically.  So, it may not be getting updated for various reasons

douglas, 2018-10-09:
It does not work when the build does not succeed 😢

douglas, 2018-10-09:
`As for the lib, it's a git submodule, specifically. So, it may not be getting updated for various reasons` it makes sense.

But I was digging on `ugfx` library [website](https://ugfx.io/pricing) to learn more about licensing and seems it can be used freely for `education/home/hobby` which seems what we do, no?

douglas, 2018-10-09:
I sent an email asking for clearance to use it in `QMK Configurator`, let's see how it goes:

```
Hello!

My name is Douglas and I'm currently learning how to build mechanical keyboards as a hobby and a way to learn more about electronics and programmable keyboards - also a way to reduce the chances of developing RSIs, as I'm a programmer and work +8h daily typing.

One of the most important tools that I use just to avoid typing more and wasting my already short time is the QMK Configurator, which allows us to use a graphical configurator for the opensource firmware to keyboards called QMK.

Here's the link in case you're curious: 

https://config.qmk.fm/#/ergodox_infinity/LAYOUT_ergodox

I'm writing to ask for clearance to allow QMK Configurator use uGfx as described in this issue I opened:

https://github.com/qmk/qmk_configurator/issues/201

TLDR: It is just used to interact with the micro led display used to show the QMK firmware logo, nothing else - and it does not do any other customization related to this library.

This a very important project for educators and hobbyists who are building they own keyboards to avoid RSI pain and strains for typing for long hours.

As a opensource developer I know who important it is to keep revenue coming so you can properly pay your employees and keep the business running and I wouldn't write if this project uses the library for commercial purposes.

Thanks for your time and kindness!
```

douglas, 2018-10-09:
Yay, this is the first reply:

```
Hello Douglas,

Thank you for contacting us.

We're extremely interested in support projects like QMK. After all we're just a bunch of engineers as well :)
The goal behind the µGFX licensing model is to do exactly what you mentioned: Keep some revenue coming that keeps the project itself alive but still support any open source project (and/or anything non-commercial) wherever possible.

I'm currently between two meetings so I don't currently have time to have a proper look at QMK. The summary from our side is: Please feel free to use µGFX within the scope of that project. We're more than happy to help wherever we can.
Please give me a few hours/days to get back to you.

Best regards,

Joel Bodenmann
uGFX GmbH
```

They were really kind! Of course we can wait more, but seems it will not be a problem given the scope of `QMK` and `QMK Configurator`.

Other than that, can I help somehow?

douglas, 2018-10-17:
Hello @yanfali! Confirmed that we can use `µGFX` with `qmk_configurator`!! Is there anything that I can do to help fixing the problem when building Ergodox Infinity? Thank y'all so much for the work on configurator! It is really awesome!

yanfali, 2018-10-19:
@douglas thanks. I'll have a word with Skully when he gets back. It's really more of a backend API change than anything we do on the UI. @skullydazed

douglas, 2018-10-22:
Yay! Thanks a lot, @yanfali! 👍

douglas, 2018-11-15:
Hey @skullydazed! Is there anything that I can do to help?

AlmondWhite, 2018-12-17:
Any update to this? Still receiving the same error code and unable to compile

MarkerDave, 2019-09-20:
Same here, I also got this error trying to compile for the Ergodox infinity.

yanfali, 2019-09-20:
Nope sorry, we can't build ugfx automatically because the license is incompatible with distribution.

KyleJamesWalker, 2020-11-22:
Just an FYI you can compile this within docker, it took quite a bit to get this figured out for a first time user.

Essentially just download the json from the QMK Configurator and then run the following to get the contents for the keymap.c:
``` bash
docker run -it --rm \
  -v $PWD/layout.json:/layout.json \
  qmkfm/qmk_firmware \
  qmk json2c /layout.json
```

Then once you have the keymap.c setup how you'd like you can compile with the following commands:
``` bash
docker run -it --rm \
  -v $PWD:/qmk_firmware/keyboards/ergodox_infinity/keymaps/docker-mount \
  -v $PWD/build:/build \
  qmkfm/qmk_firmware \
  bash -c "make ergodox_infinity:docker-mount; cp /qmk_firmware/ergodox_infinity_docker-mount.bin /build/left.bin"

docker run -it --rm \
  -v $PWD:/qmk_firmware/keyboards/ergodox_infinity/keymaps/docker-mount \
  -v $PWD/build:/build \
  qmkfm/qmk_firmware \
  bash -c "make ergodox_infinity:docker-mount MASTER=right; cp /qmk_firmware/ergodox_infinity_docker-mount.bin /build/right.bin"
```

Example with more details: https://github.com/KyleJamesWalker/qmk-ergodox-infinity-40-percent

focusaurus, 2020-11-22:
That's for the info! I was building through docker as well. My mistake was my submodules were out of sync when I rebuilt the docker image locally. I should stick to the officially published image in the future.

ieure, 2021-06-01:
Three years after the initial report, this issue persists.  If it can't be fixed, I think it'd make sense to remove the ergodox_infinity option entirely.

noroadsleft, 2021-06-02:
@ieure One of the collaborators fairly recently picked up some hardware with the goal of removing the Ergodox Infinity's dependency on uGFX. If that work comes to fruition QMK Configurator hopefully should be able to build firmware for this board soon after. :pray:

fauxpark, 2022-12-13:
uGFX is no more, this board builds in Configurator now.

closed 2022-12-13 by fauxpark