qmk/qmk_configurator #211: Bug: info.json upload for preview only works once

noroadsleft, 2018-10-31:
The Configurator feature that allows a user to upload a file to preview how it will render when merged (assuming no JSON-to-keyboard.h mismatches) by pressing Ctrl+Shift+I. It has had a bug of sorts for a while where it only worked once, and you had to refresh the page to use it again. I recently stumbled upon why that is.

When the user presses Ctrl+Shift+I, the HTML select element for `Keyboard` gains a `Preview info.json` option element as soon as the shortcut is used. If the user uploads a file, the preview loads as expected, but if the user cancels the upload, the element remains, but the select element remains on whichever keyboard was already selected.

Changing to the Preview option manually and then changing off of it removes the preview element, allowing the user to use the shortcut again.

Ideally, the preview option shouldn't be added unless the user actually uploads a file.

Additionally, the user should be allowed to upload multiple times without having to refresh the page, but that may be a separate issue.

yanfali, 2018-11-15:
This should be fixed now.

closed 2018-11-15 by yanfali

noroadsleft, 2018-11-15:
Confirmed as fixed.