qmk/qmk_configurator #217: Build errors due to network timeouts

moubry, 2018-11-21:
I'm currently getting an error compiling in the QMK Configurator:

```
See [build environment setup](https://docs.qmk.fm/build_environment_setup.html) then the [make instructions](https://docs.qmk.fm/make_instructions.html) for more information.
* Sending dz60:dz60_layout_all_mine with LAYOUT_all
* Received job_id: dd3d2fd7-4b59-4b08-b01e-0fc99f5c8408
* Running . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . .
* Finished:
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/urllib3/response.py", line 331, in _error_catcher
    yield
  File "/usr/local/lib/python3.5/dist-packages/urllib3/response.py", line 409, in read
    data = self._fp.read()
  File "/usr/lib/python3.5/http/client.py", line 461, in read
    s = self._safe_read(self.length)
  File "/usr/lib/python3.5/http/client.py", line 607, in _safe_read
    chunk = self.fp.read(min(amt, MAXAMOUNT))
  File "/usr/lib/python3.5/socket.py", line 576, in readinto
    return self._sock.recv_into(b)
  File "/usr/lib/python3.5/ssl.py", line 937, in recv_into
    return self.read(nbytes, buffer)
  File "/usr/lib/python3.5/ssl.py", line 799, in read
    return self._sslobj.read(len, buffer)
  File "/usr/lib/python3.5/ssl.py", line 583, in read
    v = self._sslobj.read(len, buffer)
socket.timeout: The read operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/botocore/response.py", line 78, in read
    chunk = self._raw_stream.read(amt)
  File "/usr/local/lib/python3.5/dist-packages/urllib3/response.py", line 430, in read
    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
  File "/usr/lib/python3.5/contextlib.py", line 77, in __exit__
    self.gen.throw(type, value, traceback)
  File "/usr/local/lib/python3.5/dist-packages/urllib3/response.py", line 336, in _error_catcher
    raise ReadTimeoutError(self._pool, None, 'Read timed out.')
urllib3.exceptions.ReadTimeoutError: AWSHTTPSConnectionPool(host='nyc3.digitaloceanspaces.com', port=443): Read timed out.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./qmk_compiler.py", line 158, in compile_firmware
    checkout_qmk()
  File "./qmk_commands.py", line 30, in checkout_qmk
    if not fetch_source(repo_name(QMK_GIT_URL)):
  File "./qmk_commands.py", line 88, in fetch_source
    zipfile_data = qmk_storage.get('cache/%s.zip' % repo)
  File "./qmk_storage.py", line 169, in get
    data = fd.read()
  File "/usr/local/lib/python3.5/dist-packages/botocore/response.py", line 81, in read
    raise ReadTimeoutError(endpoint_url=e.url, error=e)
botocore.exceptions.ReadTimeoutError: Read timeout on endpoint URL: "None"
```

yanfali, 2018-11-21:
@skullydazed are we out of storage? or is digital ocean/s3 down?

yanfali, 2018-11-21:
Hey @moubry I just tried it again and it seems to be working. Unfortunately it looks like you hit the API when there was a network problem with our storage provider and it seems to have corrected itself already. Please try again and re-open this bug if you are able to reproduce the problem. Thanks for reporting.

closed 2018-11-21 by yanfali