qmk/qmk_configurator #224: Volume up mapping is not working

link6155, 2018-12-09:
When I assign a key to be volume up in the configurator and then flash it onto my keyboard (Chimera Ortho), it doesn't do anything. The volume down and mute key works however and if I use the "Any" function to manually input the keycode, it works.

yanfali, 2018-12-09:
Thanks for reporting this. Looks like a keycode may have been mismapped. What code works for you in ANY?

yanfali, 2018-12-09:
@noroadsleft you want to check this out? I believe you were in there last. Thanks

noroadsleft, 2018-12-09:
Source looks okay from what I can tell so far. Let me build a keymap and flash it and see what happens. I'll reply again.

noroadsleft, 2018-12-09:
Tested and working as intended on my KC60.

@link6155 Do other keycodes (something other than volume control) work on the same switch, in that layer or others?

yanfali, 2019-02-08:
@link6155 any updates, or should we close this?

yanfali, 2019-02-11:
@link6155 feel free to re-open, but we're going to close this for now.

closed 2019-02-11 by yanfali