qmk/qmk_configurator #228: default keybindings seem to be broken for redox

ffinkdevs, 2018-12-24:
when i load the default keymap in the qmk configurator and compile it and flash it via qmk toolbox, all keys on the redox are wrong, but when i download  the default keymap from the qmk toolbox and flash that, its normal

ffinkdevs, 2018-12-24:
[image]
there also seems to be a big file size difference between the default in qmk toolbox and the default in qmk configurator

yanfali, 2018-12-24:
@mattdibi can you take a look please? I believe you provided the default map. @ffinkdevs I don't have a redox so I can verify this myself, but I'll ask on the QMK discord and get back to you. Binary sizes look comparable and can be explained by compiler version differences.

noroadsleft, 2018-12-24:
I think this might be my fault. I added the Configurator visual support for the Redox about six months ago, but looking at it now, I think some of the keys are out-of-sequence.

noroadsleft, 2018-12-24:
@ffinkdevs 

When the above PR is merged, your issue should mostly go away. I say "mostly" because in the process of building the above fix, I discovered that the keymap that loads in Configurator when you hit the Load Default button doesn't match the default keymap that's in the [qmk_firmware repo](/qmk/qmk_firmware/tree/master/keyboards/redox/keymap/default/keymap.c).

I also don't remember where QMK Toolbox grabs its keymap files from, or from where those files are generated.

mattdibi, 2018-12-25:
Hi, sorry for the late response.

@noroadsleft As you said the default keymap that loads the configurator doesn't match the default keymap of the qmk repo because a few, more convoluted, key-bindings were missing (namely the `LT()` macro). I had to adapt the keymap a little.

P.s. Happy holidays!

noroadsleft, 2018-12-25:
@mattdibi For the record, `LT()` is supported. Configurator will detect it as an `ANY` key. Also, the layer for the hold action has to be numeric -- `LT(1, KC_HOME)` as opposed to `LT(_NAV, KC_HOME)`.

mattdibi, 2018-12-25:
@noroadsleft Thanks for the info, I'll update the keymap when I have some time.

ffinkdevs, 2018-12-25:
> PR is merged, so it should work now.

https://config.qmk.fm/#/redox/rev1/LAYOUT
still seems to be broken

ffinkdevs, 2018-12-25:
```json
"["KC_GRV","KC_1","KC_2","KC_3","KC_4","KC_5","TT(1)","TT(2)","KC_6","KC_7","KC_8","KC_9","KC_0","KC_MINS","KC_TAB","KC_Q","KC_W","KC_E","KC_R","KC_T","KC_LBRC","KC_RBRC","KC_Y","KC_U","KC_I","KC_O","KC_P","KC_EQL","KC_ESC","KC_A","KC_S","KC_D","KC_F","KC_G","KC_PGUP","KC_END","KC_H","KC_J","KC_K","KC_L","KC_SCLN","KC_QUOT","KC_LSFT","KC_Z","KC_X","KC_C","KC_V","KC_B","KC_PGDN","KC_HOME","KC_N","KC_M","KC_COMM","KC_DOT","KC_BSLS","KC_RSFT","KC_LGUI","KC_PPLS","KC_PMNS","LALT_T(KC_PAST)","LCTL(KC_PSLS)","KC_BSPC","KC_DEL","KC_ENT","KC_SPC","KC_RALT","KC_LEFT","KC_DOWN","KC_UP","KC_RGHT"]"
```
its wrong between kc_5 and kc_6

according to https://github.com/qmk/qmk_firmware/blob/master/keyboards/redox/keymaps/default/keymap.c
the layer1 and layer2 keys have to be in the row below

noroadsleft, 2018-12-25:
@skullydazed kicked the update bot (Lord, bless him), so once &uarr; that &uarr; PR gets merged this should be fixed. (IIRC, changes to qmk_configurator/keymaps update in real time, and it's just changes in qmk_firmware that are glitchy.)

ffinkdevs, 2018-12-25:
assuming it has been integrated,
left side of the keyboard the keys are now correct, right side is "completly" broken
some keys do:
"'-098"
"=po"
"';l"
"uz]uz"
"jh"
"mn"
the rest either does nothing or i havent figured out which keys they press atm

yanfali, 2018-12-25:
Honestly the only person who can fix this is someone with the keyboard

gallexme, 2018-12-26:
@mattdibi after a long troubleshooting session discord yesterday, it was because I haven't flashed both side of the keyboards(was nowhere documented), now it works

closed 2018-12-26 by ffinkdevs