qmk/qmk_configurator #232: Download improvements

skullydazed, 2019-01-14:
We had an issue today where S3 decided to set the timeout for download keys to a very low or invalid value. When users encountered this error their browser page would change, losing configurator state. 

One possible solution is to open downloads in a new tab. Experimentation with chrome shows that it will immediately close the tab if the file is a download.

yanfali, 2019-01-14:
The only real problem is that it seems to trigger a pop up blocker. Hopefully this won't be the case when the API is working properly. I think for a binary stream the behavior is simply to save a file, so this is likely an issue where it's really content only. It's still better than losing their entire state.

zekth, 2019-06-02:
Why not using `window.open('url')` on the click? It will not be blocked if the opening is triggered by a user action.