qmk/qmk_configurator #240: Per Keyboard Keycode Sections

germ, 2019-01-26:
Now that the keyboard overlay scrolls nicely I think we can explore enabling/disabling various key sections on a per keyboard basis. I'm thinking since the keycodes are nicely organized in the docs as it is, perhaps those can be the main sections.

I propose a extra field added to info.json for any arguments passed off to config. This way existing boards are backwards compatible. For keycodes this would augment the default listing

configOpts: {
    "enabledKC": {"Steno", "OneShotKeys", "ThermalPrinter"},
    "blacklistedKC": {"RGBLighting", "MouseKeys"},
    "otherConfigOpts": ......
}

Having a config panel with a list of all options would be nice for generic use. My webdev is extremely rusty so I'm looking for some info on how to implement this in a clean nonconfusing way. My gut tells me that we should do the per keyboard level so the interface remains unchanged if needed. 

I'm working on a Steno board right now, and would love to point people towards config.qmk.fm. I think we finally can make this happen. Stenographers aren't necessarily techheads :)

Thoughts?

yanfali, 2019-01-26:
@germ let me talk with some of the other QMK folks. It's probably going to take a few days to get our thoughts organized and then we'll get back to you. Thanks, this is a pretty good idea, just not sure this is quite the right form for it.

yanfali, 2019-02-20:
@skullydazed how would you like to tackle adding extra flags to info.json?