qmk/qmk_configurator #247: Error when compiling Preonic/rev3 1x2uC

brennansang, 2019-02-11:
Hi there, when using config.qmk.fm, I get the following error whenever trying to compile for the preonic/rev3 1x2uC. I've tried compiling the default with no changes and get the same error.

> "LAYOUT_preonic_1x2uC" passed 60 arguments, but takes just 59

Please find the log below, and thanks for your great work on this!

`Preonic
===

[image]

A compact 50% (12x5) ortholinear keyboard kit made and sold by OLKB and Massdrop. [More info on qmk.fm](http://qmk.fm/preonic/)

Keyboard Maintainer: [Jack Humbert](https://github.com/jackhumbert)  
Hardware Supported: Preonic PCB rev1, Teensy 2.0  
Hardware Availability: [OLKB.com](https://olkb.com/preonic/), [Massdrop](https://www.massdrop.com/buy/preonic-mechanical-keyboard?mode=guest_open)

Make example for this keyboard (after setting up your build environment):

    make preonic:default

See [build environment setup](https://docs.qmk.fm/#/getting_started_build_tools) then the [make instructions](https://docs.qmk.fm/#/getting_started_make_guide) for more information.
Loaded 6 layers and 360 keycodes. Defined 0 Any key keycodes
* Sending preonic/rev3:defaultish with LAYOUT_preonic_1x2uC
* Received job_id: 50ae0aa8-39ab-48e8-b360-e83295dd8a6d
* Queueing . . . . . . . . . . . . . . . . . . . . . . . . . . . .
* Running . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . .
* Finished:
QMK Firmware 0.6.273
WARNING:
 Some git sub-modules are out of date or modified, please consider running:
 make git-submodule
 You can ignore this warning if you are not compiling any ChibiOS keyboards,
 or if you have modified the ChibiOS libraries yourself. 

Making preonic/rev3 with keymap defaultish

arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/preonic/rev3/matrix.c                                                          [OK]      
Compiling: keyboards/preonic/preonic.c                                                              [OK]      
Compiling: keyboards/preonic/rev3/rev3.c                                                            [OK]      
Compiling: keyboards/preonic/keymaps/defaultish/keymap.c                                           keyboards/preonic/keymaps/defaultish/keymap.c:4:445: error: macro "LAYOUT_preonic_1x2uC" passed 60 arguments, but takes just 59
  O(4), KC_LEFT, KC_DOWN, KC_UP, KC_RGHT),
                                                                                                                                                                                                                                                                                                                                                                                                                                                             ^
keyboards/preonic/keymaps/defaultish/keymap.c:4:8: error: 'LAYOUT_preonic_1x2uC' undeclared here (not in a function)
  O(4), KC_LEFT, KC_DOWN, KC_UP, KC_RGHT),
        ^
keyboards/preonic/keymaps/defaultish/keymap.c:5:445: error: macro "LAYOUT_preonic_1x2uC" passed 60 arguments, but takes just 59
  O(4), KC_LEFT, KC_DOWN, KC_UP, KC_RGHT),
                                                                                                                                                                                                                                                                                                                                                                                                                                                             ^
keyboards/preonic/keymaps/defaultish/keymap.c:6:445: error: macro "LAYOUT_preonic_1x2uC" passed 60 arguments, but takes just 59
  O(4), KC_LEFT, KC_DOWN, KC_UP, KC_RGHT),
                                                                                                                                                                                                                                                                                                                                                                                                                                                             ^
keyboards/preonic/keymaps/defaultish/keymap.c:7:543: error: macro "LAYOUT_preonic_1x2uC" passed 60 arguments, but takes just 59
  PLY),
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               ^
keyboards/preonic/keymaps/defaultish/keymap.c:8:481: error: macro "LAYOUT_preonic_1x2uC" passed 60 arguments, but takes just 59
  PLY),
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 ^
keyboards/preonic/keymaps/defaultish/keymap.c:9:524: error: macro "LAYOUT_preonic_1x2uC" passed 60 arguments, but takes just 59
  I_OFF, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS)
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                            ^
 [ERRORS]  
 | 
 | 
 | 
tmk_core/rules.mk:376: recipe for target '.build/obj_preonic_rev3_defaultish/keyboards/preonic/keymaps/defaultish/keymap.o' failed
makeap.o] Error 1
Makefile:535: recipe for target 'preonic/rev3:defaultish' failed
make: *** [preonic/rev3:defaultish] Error 1
Make finished with errors
`

yanfali, 2019-02-11:
Thank you for report. It should be fixed by this commit. 50eb6188ad6b1de11a443448bcd5515f87aef880 You will probably have to recreate your keymap by loading default again and customizing it.

yanfali, 2019-02-11:
@noroadsleft thanks for help

closed 2019-02-11 by yanfali