qmk/qmk_configurator #255: Frontend needs a way to remove layers after they are generated

BlueNalgene, 2019-02-20:
When using, I was typing in the MO box that a key should be for layer 1.  I did not highlight the 0 at first, so it thought I wanted layer 10, and it dutifully enabled it for me.  After changing it back to 1, the layer for 10 was still visible.  Although there were no keycodes mapped to that layer.

This caused compilation failure.  The code decided that since I included layer 0, 1, 2, and "10", that I needed the layers 3-9.  It generated them in the code as:

```
        [3] = LAYOUT_all(),
	[4] = LAYOUT_all(),
	[5] = LAYOUT_all(),
	[6] = LAYOUT_all(),
	[7] = LAYOUT_all(),
	[8] = LAYOUT_all(),
	[9] = LAYOUT_all(),
```

The empty LAYOUT_all() calls triggered an error:

```
keyboards/dz60/keymaps/foobar/keymap.c:7:19: error: macro "LAYOUT_all" requires 68 arguments, but only 1 given
  [3] = LAYOUT_all(),
                   ^
keyboards/dz60/keymaps/foobar/keymap.c:7:8: error: ‘LAYOUT_all’ undeclared here (not in a function)
  [3] = LAYOUT_all(),
        ^
```

My guess is that this is somewhere in the JS code (which I don't speak, and can't help fix) to script the keymap generation.  Unless others experience the same issue, it may be low priority.  The keymap is still generated and exportable, even if the hex is not.  I can just compile that myself.

[JSON file (saved as txt) for reference)](https://github.com/qmk/qmk_configurator/files/2886184/foobar.json.txt)

yanfali, 2019-02-20:
Thank you so much for filing this issue. This is definitely a problem and I am planning on tackling on that next, after I get LT support shipped out. There are a couple of ways to fix this, one is to simply generate empty layers between. I'm probably going to do that anyway, as the backend API seems to be generating empty layers when it shouldn't be. @skullydazed may have more insight.

The other UI change I'm planning on working is to provide a way to swap and remove layers.

yanfali, 2019-02-20:
Related to #188

BlueNalgene, 2019-02-20:
Is there any reason not to generate all of the possible layers, even if they are not being used?  If I remember my TMK code, the number of layers is capped at a finite number.  You do not save much room by omitting them.  Even then, the compiler probably ignores the transparent keys.  The only advantage is readability, which is presumably less of an issue for people using a script generator.

yanfali, 2019-02-20:
Mostly that it's a regression in the API and not an actual feature. Compiling empty layers takes up precious flash space, and on AVR platforms that can be in short supply.

yanfali, 2019-04-08:
@BlueNalgene good news, we have just implemented #287 and this should handle most of your issues. You can effectively now delete a layer by remove the keys. We insert dummy KC_TRNS layers if you choose to create a sparse layer keymap. The downside is you will have a larger binary and on some keyboards which are tight on space it may generate a hex file that is too large. Feel free to re-open this issue if this doesn't solve the problem.

closed 2019-04-08 by yanfali