qmk/qmk_configurator #260: Key shape simplified

i5ar, 2019-02-24:
I have a few suggestions regarding the key shape as defined in the [docs](https://github.com/qmk/qmk_firmware/blob/bcb18154206332c85e500ef65bffa98be6ac1cb7/docs/reference_info_json.md).

1. As you can see the first [x, y] coordinate is always equal to the last one so my first suggestion is to define the key shape as a polygon and not as a polyline so the last [x, y] coordinate can be removed;
2. I think one level array in this case is better than a nested array since the shape of a key will be generally short and therefore readable without the need to nest [x, y] pairs. This is also how polygons are defined in SVG ([`<polygon points="0, 0, 150, 0, 150, 200, 25, 200, 25, 100, 0, 100" />`](https://jsfiddle.net/80taucq1/));
3. I think the origin should not be in the top-left but in the same position used by the rectangle. This will make it easier to design the configurator since the origin is the same with or without a key shape value.

Maybe it's too late or maybe you don't like this approach but if you take this road I suggest using the key `p` (points) to avoid breaking changes with `ks`. 

In conclusion the ISO key would look like:

    "p": [-0.25, 0, 1.25, 0, 1.25, 2, 0, 2, 0, 1, -0.25, 1]

Instead of:

    "ks": [[0, 0], [1.5, 0], [1.5, 2], [0.25, 2], [0.25, 1], [0, 1], [0, 0]]

Thank you.

yanfali, 2019-02-24:
Great suggestions. So right now everything is defined as divs to facilitate drag and drop. I did experiment with svg here

https://yanfali.github.io/keycodes/

The issue being interaction with canvas vs DOM. They don't really mix all that nicely. I need to revisit the rendering algorithm using svg elements rendered inside DOM.

The coordinate system is actually in key units and is converted to pixels. We actually need to support rotation to so at some point this does need revisiting.

Thank you for the feedback, it's appreciated.

yanfali, 2019-02-24:
Please check out the discussion https://github.com/qmk/qmk_firmware/issues/2621#issuecomment-388484391 and feel free to contribute your feedback. Info.json is definitely going to evolve. It's not just used for rendering though. It's part of the API interaction between UI and qmk

i5ar, 2019-02-24:
> The coordinate system is actually in key units and is converted to pixels.

That's perfect. In my suggestion I only changed the structure of the array and the position of the relative origin from here:

```
○----+
|    |
+-+  |
  |  |
  +--+
```
to here:
```
+-○--+
|    |
+-+  |
  |  |
  +--+
```

> We actually need to support rotation to so at some point this does need revisiting.

I'm building something but I haven't tried to support rotation yet. If I have time, I'll try to implement something this night, just to make sure this approach works nicely with more complex layouts.

yanfali, 2019-02-24:
Very neat. @harshitgoel was experimenting with that a month or so ago, but I think he got busy. Where to do the rotation is a matter of contention. KLE is our reference implementation as they do currently support rotation and one of the sources of layout information is KLE. One key feature or our current implementation is drag-n-drop so it's important to not pick technologies that prevent that mode of operation.

i5ar, 2019-02-25:
I wouldn't use KLE as reference for `rx` and `ry`.
Rotation `r` works fine but `rx` and `ry` have some strange behavior.
When `rx` is defined `x` decrease the same value and the value in the *Raw data* tab is different from the value in the *Properties* tab.

You can always import data from KLE and fix this issue on your side so the API can serve the correct values for `x` and `y`.

KLE data:
```js
{
    x: 0,
    y: 0,
    r: 10,
    rx: 1,
    ry: 1
}
```

QMK info (suggestion):
```js
{
    x: 1,  // Same as `rx` when `x: 0` in KLE data
    y: 1,  // Same as `ry` when `y: 0` in KLE data
    r: 10,
    rx: 1,
    ry: 1
}
```

To prove the suggestions above work nicely I made an [example](https://jsfiddle.net/arken1mj/2/).

i5ar, 2019-02-26:
I investigated a little more into the KLE [docs](https://github.com/ijprest/keyboard-layout-editor/wiki/Serialized-Data-Format) and I can confirm that KLE is using the same origin I proposed for oddly-shaped keys, as you can see [here](http://www.keyboard-layout-editor.com/##@@_w:1.25&h:2&w2:1.5&h2:1&x2:-0.25%3B&=Enter).

Basically this one:
```
+-○--+
|    |
+-+  |
  |  |
  +--+
```

skullydazed, 2019-02-26:
`x` and `y` in KLE is a different concept. In KLE `x` and `y` default to 1 and 0 respectively and represent the offset from the previous key. In QMK those are an absolute coordinate. Statelessness is a key feature of our format, and is what drives most of our deviations from KLE. As such you can't compare them [this way](https://github.com/qmk/qmk_configurator/issues/260#issuecomment-467006479).

It's interesting that KLE draws ISO enter the way it does, and this has been convenient for our early work as iso enter has "just worked" with a 1.25x2 rectangle. I think the ISO Enter change you're proposing is simply a change to convention, not a change in implementation, and probably makes sense. We aren't using it anywhere yet AFAIK, so changing this now is pretty easy. Thanks for helping us to see this necessity of this!

I see your point (no pun intended) about collapsing point pairs into a single list rather than nested but I'm not a fan of that approach because it makes working with the format harder if you're not intimately familiar with it or svg, and most users of QMK are not experts. Plus, nesting them this way lets you use code formatting tools to break them out into lines for easy and convenient editing.

yanfali, 2019-02-26:
I actually think there's a way to do an ISO key using pure CSS using :after and :before selectors, I just haven't time to look into it.

yanfali, 2019-02-26:
@i5ar if you'd like to look at doing ISO key support in pure CSS I would be most appreciative

i5ar, 2019-02-26:
> KLE is a different concept [...] Statelessness is a key feature of our format, and is what drives most of our deviations from KLE.

@skullydazed I realized that today as I found the docs of KLE in the Wiki. As a matter of fact, I had a hard time explaining how I made the rotation works when using `rx` and `ry` (the example is above).

> It's interesting that KLE draws ISO enter the way it does, and this has been convenient for our early work as iso enter has "just worked" with a 1.25x2 rectangle.

:smile: Exactly!

> I see your point (no pun intended) about collapsing point pairs into a single list rather than nested but I'm not a fan of that approach because it makes working with the format harder if you're not intimately familiar with it or svg, and most users of QMK are not experts.

About collapsing point pairs into a single list:
- Even indexes are `x` values, odd indexes are `y` values, so it's still pretty easy to work with;
- The array is smaller;
- Most developers only need to copy the array without editing;
- This convention is used to draw polygons not only in SVG but also in [HTML5 Canvas](https://stackoverflow.com/questions/4839993/how-to-draw-polygons-on-an-html5-canvas).

Also, some JSON formatter tools do this boring expansion:
```js

          "ks": [
            [
              0,
              0
            ],
            [
              1.5,
              0
            ],
            [
              1.5,
              2
            ],
            [
              0.25,
              2
            ],
            [
              0.25,
              1
            ],
            [
              0,
              1
            ],
            [
              0,
              0
            ]
          ],
```
Just something to consider that I wanted to point out.
Thank you.

i5ar, 2019-02-26:
@yanfali This should work.

```html
<div id="enter"></div>

<style>
#enter {
  background: teal;
  height: 200px;
  position: relative;
  width: 125px;
}
#enter:after {
  background: teal;
  content: "";
  height: 100px;
  left: -25px;
  top: 0;
  position: absolute;
  width: 25px;
}
</style>
```
Of course, you need to scale the pixels accordingly with the configurator.

yanfali, 2019-02-26:
Awesome, thanks. I'll take a look when I get a chance. I may be a few days.

yanfali, 2019-04-08:
@i5ar I gave it a shot, but it didn't really work. I'm thinking that we may need to use 3 divs to represent this one key with the appropriate coloring and styling.

i5ar, 2019-04-08:
@yanfali This is how much further I could go https://jsfiddle.net/yzq2m6pc/3/ from here https://jsfiddle.net/zqg78yeb/