qmk/qmk_configurator #287: Suggestion: new layers should only be saved if something is added to it

1hella, 2019-03-17:
I had the problem of a bunch of empty layers being created by clicking on them, which in my case gave an error during compilation.

I think just clicking a layer shouldn't save it. I suggest that if somebody clicks on a new layer, it should only be saved after a key is changed on it. In other words, when clicking away from a layer, if every key code is KC_NO, delete it.

Alternatively, having a button to delete an accidentally created layer would work as well.

yanfali, 2019-03-17:
Yep thanks for raising issue. It's been on my list to add layer editing for a while so I will be working on this in the future.

yanfali, 2019-04-08:
@1hella give the latest configurator a try. It should be working the way you expect. Thanks for the suggestion. We also support sparse layers, but the downside is the keymap generated will be larger, so this may cause problems for some keyboards.

closed 2019-04-08 by yanfali