qmk/qmk_configurator #298: Support number pad and function keys

i5ar, 2019-03-25:
A few suggestions:
- Support number pad keys using `event.location` as you already did with left/right modifiers https://github.com/qmk/qmk_configurator/commit/55f6df70bea9ef7a7a5dc1165d73a7966cd4adca.
- Support function keys:
  - Add the `event.key` value to `/modules/keycodes/ansi.js` (e.g.  `keys: "F1"`);
  - Intercept the browser function keys (i.e. https://keycode.info).

Thank you.

i5ar, 2019-04-04:
To disable browser-specific shortcuts (e.g. `F5`) the `event.preventDefault()` in the key down handler should be enough. If the handler is inside the keyboard component, it'll work only when the focus is on the selected key and we can still use browser-specific shortcuts when the focus is elsewhere.

Regarding the support for number pad keys, it's slightly more complicated and it may require to change the data structure of the keymaps to someting like this:
```js
  {
    name: "0",
    code: "KC_P0",
    keys: ["Insert", "0"],  // type: Array
    location: 3             // type: int
  },
```

yanfali, 2019-04-04:
I kind of agree on the keypresses, that can probably be made to work. I think numpad can just work using location so your idea is probably unnecessary. I look forward to your PR.

i5ar, 2019-04-04:
> I think numpad can just work using location so your idea is probably unnecessary.

I think `keys` should be an array because "NumLock" can be toggled: The `event.key` can be either "Insert" or "0", "End" or "1" and so on.

Of course you can do the other way around if you map elsewhere "Insert" to "0", "End" to "1", etc.

yanfali, 2019-04-04:
I need to test this, but the idea is if you get a 0 and the location is numpad it is on. You wouldn't get a zero otherwise, you would get an insert. We bind to the output, not the keycode because that's what keypress.js cares about.

i5ar, 2019-04-05:
In another project, the structure I suggested above let me easily handle any kind of keypress (numpad, caps, modifiers) while keeping the logic slim without using 3rd-party libraries.
I just wanted to point that out. After all, "KC_A" can be either "a" or "A" at the same location so it makes sense for `keys` being an array.

Your approach requires fewer changes and I'm sure it works well too.

yanfali, 2019-04-05:
Actually that's not true. The os only received a single code. It's affected by the state of caps lock.

i5ar, 2019-04-05:
> Actually that's not true. The os only received a single code.

I was not talking about the os. I was talking about the `event.key`, documented [here](https://developer.mozilla.org/en-US/docs/Web/API/KeyboardEvent/key), that you need to handle to set the appropriate "KC_*".
You want both `key='A'` and `key='a'` to target "KC_A". The library that you are using seems to solve already this problem for the alphanumeric keys but not for all the keys.
As I said, your solution works and it requires fewer changes. You just have to write more if/else  (especially if you want to support 'AltGraph' too).

yanfali, 2019-04-05:
Fair. Thanks for the input.

PR #371 (2019-05-25): support number pad and function keys #298
- detects number pad keys following the pattern used for mods in https://github.com/qmk/qmk_configurator/commit/55f6df70bea9ef7a7a5dc1165d73a7966cd4adca
- adds function row support, and prevents default behavior using `preventDefault` (only on function keys)

closes #298
closed 2019-05-25 by yanfali