qmk/qmk_configurator #319: Feature: generate printable layout overview

yourfate, 2019-04-23:
The configurator can already display all the layer keymaps. Would be cool if there was a button to download a PDF where all the layers are displayed. Great for pinning next to your screen to learn your new layout / look up rarely used keys on obscure layers. Right now I create this myself by screenshotting the different layers...

yanfali, 2019-04-23:
We've actually discussed this in the past. Now that we just added i18n I think we think about adding this. I would need to generalize the visual rendering component a bit more as right now it's tied to the needs of interactive UI. It would probably be easier to create a stripped down base rendering component that was shared. Probably won't do a PDF, but a simple HTML page with a printer style sheet and then you can use print to PDF to get it in that format. I don't want to deal with PDF in a browser.

yanfali, 2019-05-10:
There's a branch where I work on this as time permits.

closed 2019-05-23 by yanfali