qmk/qmk_configurator #376: Character being dropped from keymap on compile (online configurator)

Sushiimi, 2019-05-27:
Was compiling my firmware using the online configurator and it seems as though a character is being dropped from the keymap for some reason causing the compilation to fail. However, the downloaded layout JSON file does not have that removed character.  

* PCB/Board: `waldo`
* Layout: `LAYOUT_60_HHKB`

Here is the output from the console on the configurator page:
```Compiling: keyboards/waldo/waldo.c                                                                  [OK]      
Compiling: keyboards/waldo/keymaps/waldo_layout_60_hhkb_mine/keymap.c                              In file included from keyboards/waldo/keymaps/waldo_layout_60_hhkb_mine/keymap.c:1:0:
keyboards/waldo/waldo.h:32:98: error: ‘C_NO’ undeclared here (not in a function)
     { KC_NO, k41,   k42, KC_NO, KC_NO, KC_NO, k46, KC_NO, KC_NO, k49, k4a, KC_NO, KC_NO,  KC_NO, C_NO  }  \
                                                                                                  ^
keyboards/waldo/keymaps/waldo_layout_60_hhkb_mine/keymap.c:4:8: note: in expansion of macro ‘LAYOUT_60_hhkb’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_LCTL, KC_LGUI, KC_LALT, KC_SPC, KC_RALT, KC_RGUI),
        ^
 [ERRORS]  
 | 
 | 
 | 
tmk_core/rules.mk:369: recipe for target '.build/obj_waldo_waldo_layout_60_hhkb_mine/keyboards/waldo/keymaps/waldo_layout_60_hhkb_mine/keymap.o' failed
makeap.o] Error 1
Makefile:535: recipe for target 'waldo:waldo_layout_60_hhkb_mine' failed
make: *** ine] Error 1
Make finished with errors```

Have not tested this with out boards or layouts for the same PCB but can help in doing so if need be! Let me know if there is anything I can do to help fix the issue and I would be happy to help!

noroadsleft, 2019-05-27:
Nice find!

This is an error in the codebase; you did nothing wrong.

https://github.com/qmk/qmk_firmware/blob/6c7c178cd2b9d535e76a0cd9dd3d1293b8ba1ed2/keyboards/waldo/waldo.h#L32

I'll open a pull request; it's literally a one-character fix.

Sushiimi, 2019-05-27:
Yup, that is what I figured! Feel free to ping me to test it when the changes are live and I can report back 😄

noroadsleft, 2019-05-28:
@Sushiimi This should be fixed now.

Sushiimi, 2019-05-28:
@noroadsleft Yup, all working now! :+1: Appreciate it!

noroadsleft, 2019-05-28:
Thanks for the report!

closed 2019-05-28 by noroadsleft