qmk/qmk_configurator #377: add the options for config.h

daenorks, 2019-05-28:
Hello, i would like to know if it was possible to add the possibility to customize the config.h file and is parameters in an option window or something...

https://beta.docs.qmk.fm/reference/config_options

And is it possible to make a donation ?

noroadsleft, 2019-05-28:
Hello,

Editing `config.h` isn't currently possible in the Configurator, because the API that supports its back end doesn't have any endpoints for doing so at this time.

As for supporting, the ways you can support QMK are listed at https://qmk.fm/support/.

Thanks!

yanfali, 2019-05-28:
@skullydazed fyi

rmwphd, 2020-01-13:
> Editing `config.h` isn't currently possible in the Configurator, because the API that supports its back end doesn't have any endpoints for doing so at this time.

Is this API part of this repo? Or where could I find it to add such endpoints?

yanfali, 2020-03-03:
@rmwphd you should look at the qmk-cli effort, that's where this feature would fit best.

karl-run, 2021-04-02:
If anyone like me comes across this issue from a google search like me. Here's what I realized:

When you set up the QMK CLI, you are able to go into `qmk_firmware/keyboards/your-keyboard/config.h` and add your config values there. When you are running `qmk compile -c your-json-from-configurator.json` or `qmk flash-c your-json-from-configurator.json` it will compile the firmware from this source with your configurations.

In my case I wanted to set `IGNORE_MOD_TAP_INTERRUPT` for my Redox, so adding `#define IGNORE_MOD_TAP_INTERRUPT` to `qmk_firmware/keyboards/redox/config.h` and ran `qmk complie` it worked!

Might be obvious to some, but I'm pretty new to QMK so it took some time for me to understand how the CLI worked compared to the QMK Configurator + QMK Toolbox for flashing.

skullydazed, 2021-04-02:
Glad you were able to figure that out @karl-run!

If you want to use more than one keyboard in the future here's something that builds on what you learned- you can use your configurator export instead of `keymap.c` by doing this:

* `cd qmk_firmware/keyboards/your-keyboard/keymaps/`
* `mkdir your-keymap`
* `cd your-keymap`
* `cp ~/your-json-from-configurator.json keymap.json`
* `echo #define IGNORE_MOD_TAP_INTERRUPT` > config.h
* `qmk compile -c`

That will let you override settings without changing the global keyboard settings. Really handy if you have more than one keyboard.