qmk/qmk_configurator #380: Generated .bin file for Instant60 cause the keyboard to hang

bengine, 2019-05-31:
Using QMK Configurator to compile and generate .bin file for Instant60.

However, the generated bin file caused the keyboard to be unresponsive once the keyboard is flashed with it.

I do not encountered such issue when I compile the .bin file locally.

I also noticed that the .bin file generated by QMK Configurator to be 1kb bigger than local compile.

[INSTANT60.zip](https://github.com/qmk/qmk_configurator/files/3240105/INSTANT60.zip)

Attached file contains the .json & .bin file generated by QMK Configurator, as well as .bin file that I compiled locally.

bengine, 2019-05-31:
I also tried the .bin file from the https://qmk.fm/keyboards/. It also DOES NOT work.

drashna, 2019-05-31:
For reference: 

Travis CI (which compiles the image at qmk.fm/keyboards) uses arm-none-eabi-gcc version 4.9.2.

bengine, 2019-05-31:
I'm using arm-none-eabi-gcc version 6.3.1 locally.

yanfali, 2019-05-31:
@bengine thanks for doing this, we'll look into getting the compilers upgraded and ping you when we have a fix. ARM is the future of QMK so we're motivated to fix this too.

yanfali, 2019-06-02:
We're working on a fix here https://github.com/qmk/qmk_firmware/pull/6043

yanfali, 2019-06-06:
Should be fixed now, on both qmk.fm/keyboards and configurator. Thanks for your patience.

closed 2019-06-06 by yanfali