qmk/qmk_configurator #404: Don't track keyup for capslock

yanfali, 2019-06-08:
The OS doesn't ever seem to send a keyup for capslock. so just register that it was hit and don't try and track it ever going up.

zekth, 2019-06-10:
Can't reproduce on W10.

yanfali, 2019-06-10:
Good to know, must be an osx thing

yanfali, 2019-06-10:
Oh, my caps key is on a layer...

zekth, 2019-06-11:
this issue can be closed so?

yanfali, 2019-06-11:
I don't know. It's kind of annoying on mac, but I'm not a big caps lock user :D

zekth, 2019-06-11:
> 
> 
> I don't know. It's kind of annoying on mac, but I'm not a big caps lock user :D

Will test on my rotten apple env.

zekth, 2019-06-12:
This is related to MacOs. Activating CapsLock sends a key-down, deactivating it sends a key-up.

yanfali, 2019-06-13:
crazy right? should we just ignore it?

zekth, 2019-06-15:
I'd personnaly ignore it. But is it worth it to detect key-up / Key-down? Maybe just refactor it to key-event and make a fallback for this special case?

yanfali, 2019-07-03:
It's probably related to OSX requiring a minimum delay before recognizing a caps lock. Closing

closed 2019-07-03 by yanfali

akellbl4, 2021-02-08:
Hi folks!
@zekth I'm using caps lock as an input source switcher (switch between different languages) on macOS. 
I have a problem because caps lock doesn't send "key up" event. It's an annoying thing because if I switch to another language caps lock becomes turned on.