qmk/qmk_configurator #413: Allow environment variable override for backend_baseurl

cscheib, 2019-06-19:
it's OK that we hardcode backend_baseurl as a constant, but we need a way to override it during development without having to manually edit the config file.

This will be particularly useful for the docker-compose web stack at https://github.com/qmk/qmk_web_stack

cscheib, 2019-06-19:
https://github.com/qmk/qmk_configurator/blob/e84633a9cc8489da8095d587618d2520b0bdc601/src/store/modules/constants.js#L3

yanfali, 2019-06-19:
well a javascripty way to deal with this is take advantage of process.env.NODE_ENV. Check it for production and use one value, for anything else, just use an environment variable if it exists.

yanfali, 2019-06-19:
@zekth you want to take a stab at this one?

cscheib, 2019-06-20:
@yanfali I may take a crack at it myself, once I finish with a couple other tasks I'm working on.  Whoever gets to it first.

I'm not sure I'd interpret NODE_ENV for this one -- there may be some valid reason for someone to run one of these of their own, so why not just default to api.qmk.fm (as-is) and if-and-only-if the environment variable exists, use the value of that.  I personally believe NODE_ENV tends to be abused by node devs

yanfali, 2019-06-20:
@cscheib we mostly use NODE_ENV for build time stuff, there's some code to selectively replace files for google analytics we aren't using anymore that would do what you are after. However since this is actually a dev only feature, environment variables are appropriate.

PR #518 (2019-10-04): add env variable handling for API url
fix #413 
Based on the vue cli constaints : https://cli.vuejs.org/guide/mode-and-env.html#modes

.env file is loaded and can be modified for a local development environment.
.env.production is used when we use `yarn build`

Also if the runtime doesn't find the variable, it throw exception at startup.
closed 2019-10-04 by yanfali