qmk/qmk_configurator #424: keep list of user's favorite keyboards in local storage and filter UI

yanfali, 2019-07-03:
Users would be able to pick which keyboards they prefer to see in their drop down, and toggle this on and off. Maybe have a :heart: icon which a user would click to add it to list and a :filter: icon to toggle the feature on and off.

Because all data is stored in the browser and never sent anywhere else this should not contravene GDPR because there is no PII (Personally Identifiable Information) beyond a simple list of keyboards in local storage, and possibly the layout.

zekth, 2019-07-05:
This may be done after the refactor involved by https://github.com/qmk/qmk_configurator/issues/415 ?

yanfali, 2019-07-05:
That's probably a good idea.

closed 2019-08-25 by yanfali