qmk/qmk_configurator #435: Sounds and Music Mode unavailable after flashing

GrinninBarrett, 2019-07-14:
I just flashed my new Planck Rev6 (on Mac) using QMK Configurator,  with very simple changes: adding a right shift on the base layer where "Enter" is by default, and using "Lower+Enter" to actually hit "Enter". I also added "DEL" to the "Backspace" key in both "Lower" and "Raise" layers. I didn't change anything else.  After flashing, music mode is unavailable, and the only sounds the keyboard will make are when plugging it in. I re-flashed the default keymap in attempt to undo that, but the music mode is still gone. How can I fix this?

noroadsleft, 2019-07-14:
Keymaps generated and compiled by QMK Configurator really only include the code necessary to use your keyboard as a keyboard, as the API that supports Configurator doesn't currently have the ability to make source-level changes like enabling/disabling options like Audio or RGB Underglow.

The only way to get these features back would be to set up a development environment and compile your keymap from source code. For instructions on how to do this, please see the QMK Docs:

- [Getting Started - QMK Firmware](https://docs.qmk.fm/#/newbs_getting_started)

If you need any help with this just reply back.

GrinninBarrett, 2019-07-15:
Thank you for the reply. Is it possible then to make the changes I made,
but return the board to its default condition otherwise (including
reintroducing audio functionality)? I’m not particularly fluent with code
but have a basic understanding of C. I could probably figure it out but
would be afraid to mess something else up.

Thanks,
Tucker

noroadsleft, 2019-07-15:
Absolutely it's possible. The Configurator creates a stripped-down `keymap.c` file before it compiles your firmware.

When you get the development environment set up, go to `keyboards/planck/keymaps/` and make a copy of the `default` folder, and name the new folder `grinninbarrett` (or whatever suits you). Then, open the `grinninbarrett/keymap.c` file in a plain-text editor. Starting at Line 57, you'll see:

```
[_QWERTY] = LAYOUT_planck_grid(
    KC_TAB,  KC_Q,    KC_W,    KC_E,    KC_R,    KC_T,    KC_Y,    KC_U,    KC_I,    KC_O,    KC_P,    KC_BSPC,
    KC_ESC,  KC_A,    KC_S,    KC_D,    KC_F,    KC_G,    KC_H,    KC_J,    KC_K,    KC_L,    KC_SCLN, KC_QUOT,
    KC_LSFT, KC_Z,    KC_X,    KC_C,    KC_V,    KC_B,    KC_N,    KC_M,    KC_COMM, KC_DOT,  KC_SLSH, KC_ENT ,
    BACKLIT, KC_LCTL, KC_LALT, KC_LGUI, LOWER,   KC_SPC,  KC_SPC,  RAISE,   KC_LEFT, KC_DOWN, KC_UP,   KC_RGHT
),
```

Change `KC_ENT` to `KC_RSFT`, then jump down to Line 114, and change the last `_______` to `KC_ENT`. Then on Lines 112 and 130, change `KC_BSPC` to `KC_DEL`. Save the file, and then run your command line interface, and `cd` to your `qmk_firmware` folder, then run `make planck/rev6:grinninbarrett` to compile your firmware.

When `make` finishes compiling your firmware, you should see a line that says something like "Firmware size check does not work on Cortex processors, skipping." At this point, you can use QMK Toolbox to install your new firmware to your Planck. The file will be in your `qmk_firmware` folder, with the filename `planck_rev6_grinninbarrett.bin`. From here flash the file to your Planck just like you've done previously.

drashna, 2019-07-15:
Actually, I suspect that you turned off audio completely, which would prevent music mode from working.

Make sure you have `AU_ON` (Audio On) in your keymap, and see if that helps

GrinninBarrett, 2019-07-15:
Hello,

Well of course I hope it's possible to return the board to its "factory"
condition, including sounds. I would be really bummed if that feature is
gone forever, even though it's a very small part of the reason I love the
board. I will try to flash it again, but using the actual code as you
suggested. So, will I just need to copy and paste the code from the default
keymap as found on the QMK website? Then make the changes as you mentioned?
Is it basically that straightforward?

Thanks,
Tucker

skullydazed, 2019-07-15:
You can always flash the latest version of the factory firmware by downloading it from here: 

https://qmk.fm/compiled/planck_rev6_default.bin

That is linked from the master list of keyboards on qmk.fm:

https://qmk.fm/keyboards/

GrinninBarrett, 2019-07-15:
This worked perfectly! It's exactly what I needed. I shouldn't have been so
daunted by the prospect of using actual code before; it was simpler than I
expected, thanks to your guidance. Thank you so much for your help!

noroadsleft, 2019-07-16:
Glad you got it sorted. I'm going to close this issue now, but feel free to reopen it if you need to do so.

closed 2019-07-16 by noroadsleft