qmk/qmk_configurator #439: Keebio

emanpatricio, 2019-07-23:


emanpatricio, 2019-07-23:
What am I doing wrong the HJKL, when configured as an arrow for layer 1, it's not working as expected. 

this is the config
https://imgur.com/kNTbpcr

noroadsleft, 2019-07-23:
That should work as long as you're holding down the `MO(1)` key while attempting to use the HJKL arrows.

emanpatricio, 2019-08-02:
Sorry for the late replied, I really don't know what I'm doing wrong, but I had an opportunity to try again this afternoon, the Numpad on MO(1) is not getting overwritten.

Tonight I will try with my rev3 keebio hopefully, it will work.

yanfali, 2019-08-02:
@emanpatricio in your screen shot you'd selected the iris v2. That has an incompatible matrix with the rev3 so it probably won't work. Please try again with a rev3 

https://config.qmk.fm/#/keebio/iris/rev3/LAYOUT

yanfali, 2019-08-16:
No response in 14 days. Closing. Please reopen if this is still an issue.

closed 2019-08-16 by yanfali