qmk/qmk_configurator #458: Feature: Option to download .hex file

haaland, 2019-08-06:
Planck v6 download firmware only gives me the .bin file, would like to choose .hex.

skullydazed, 2019-08-06:
Do you have a tool that will flash hex files on stm32? AFAIK there are only tools to flash .bin files.

yanfali, 2019-08-06:
@haaland http://www.keil.com/download/docs/113.asp

yanfali, 2019-08-06:
@haaland As @skullydazed mentioned, we are not aware of any ARM based keyboards that use hex files at this time. If you know of a product that requires a hex file please let us know. As have to do something similar for the tada68 which requires a bin file for the default bootloader because it uses USB Storage. Please feel free to close this issue if you think it has been resolved.

yanfali, 2019-08-16:
No response for 10 days. Closing.

closed 2019-08-16 by yanfali