qmk/qmk_configurator #468: encoder support [bdn9]

PeterHindes, 2019-08-16:
The configurator seems to have no way to configure [encoders](https://beta.docs.qmk.fm/features/feature_encoders), and even the default layout for the bdn9 when compiled with QMK configurator has no mappings for the encoders.,

PeterHindes, 2019-08-16:
[defaults.zip](https://github.com/qmk/qmk_configurator/files/3508170/defaults.zip)

yanfali, 2019-08-16:
@PeterHindes yeah, sorry, but there is no support in the API right now to do this. The idea behind downloading source is you can use the configurator to build a starter than then use the full environment to create an encoder mapping. Currently this is an advanced feature and there's no way for the UI to access it with the current API. This will hopefully change in the future when we have a data driven code generator. If you want to help with that, please join the discord.

noroadsleft, 2019-08-16:
> please join the discord.

which is here: https://discord.gg/Uq7gcHh

xplusplus, 2020-03-16:
I'm actually working on a POC for this with the CLI right now: #8440