qmk/qmk_configurator #479: Keycodes for Ctrl and OS are swapped.

AffectionateNouns, 2019-08-22:
I don't know for sure if it's just for the DZ65RGB or if it exists on other boards, but the L_CRTL and L_OS keycodes are swapped, as well as R_CTRL and R_OS. Meaning if you select L_CTRL for the bottom leftmost key on the board (where Ctrl would normally be) it sets that to be your OS key.

yanfali, 2019-08-22:
@noroadsleft ?

noroadsleft, 2019-08-23:
This is actually the combination of a few different issues, but the interim cause is a recent commit to QMK Firmware that exposes a long-standing weakness in how QMK Firmware uses EEPROM (meaning this isn't a Configurator bug).

The data for storing keyboard settings across power cycles (unplugging/replugging the keyboard, installing new firmware, etc.) is stored in a section of space called [EEPROM](https://en.wikipedia.org/wiki/EEPROM). Because QMK's EEPROM system isn't version controlled, changes to the EEPROM structure can lead to unexpected behavior. QMK doesn't currently have the capability to migrate settings across changes to the EEPROM structure. Because of this, when a new setting is being stored in EEPROM, the place on the keyboard's microcontroller where the new setting will be saved may already have data on it, and that data may pre-configure the new setting when QMK starts up. A recent commit added the capability to swap the Ctrl and OS keys in real time, and persist it across power cycle events (same as the Alt/OS or Caps Lock/Ctrl key swaps). The lack of version controlled EEPROM means the space where new QMK Firmware files saves the setting for this had a non-zero value in it, which made QMK believe you had enabled the Ctrl/OS swap.

The interim solution to this is to reset your EEPROM. QMK Toolbox may be able to do this (I'm not sure for the DZ65RGB), but Configurator has a keycode for this labeled `EEPROM Reset` on the Quantum tab, next to the normal `Reset`. Program this key in your keymap, and program a different key to `CG_NORM` (use the Any key feature on the Quantum tab), then compile and flash the firmware. Pressing the `EEPROM Reset` key should fix this, but if it doesn't, press the `CG_NORM` key as well.

- The "offending" pull request: qmk/qmk_firmware#6110
- Relevant Issue: qmk/qmk_firmware#6589
- Solution: https://github.com/qmk/qmk_firmware/issues/6589#issuecomment-524042457

drashna, 2019-08-23:
@stanrc85 for `EEP_RST`, did you unplug the board and plug it back in?

Also, this is for the HS60v2, do you have VIA enabled on it?   if so, that is the problem.  It's getting reset by VIA..... 

https://github.com/qmk/qmk_firmware/blob/master/keyboards/hs60/v2/config.h#L138-L144

increment the addresses by 2, and it should fix it.

stanrc85, 2019-08-23:
Yeah after doing the eep_rst I have to unplug the board. I don't use VIA but I think it is enabled on my build. I'll try to increase the address later and let you know. Thanks!

drashna, 2019-08-23:
I checked the address, it does collide. So it will need to be changed

stanrc85, 2019-08-23:
The magic or version addr or both?

drashna, 2019-08-23:
See the other thread.

stanrc85, 2019-08-24:
That resolved it, thanks!

yanfali, 2019-09-05:
ok, closing this as it's an issue with firmware and has now been patched.

closed 2019-09-05 by yanfali

resuna, 2024-08-29:
On my Y&R 6095 I had to hold down the ESC key while plugging it in to get the EEPROM reset, none of the keyboard mapping through special keys to RESET the board worked. Apparently Y&R Studios hasn't updated the firmware they ship in a while.