qmk/qmk_configurator #488: Layer and media keys not working in Windows 10

johann-koebbe, 2019-09-06:
I don't know if this is the right place for this issue, but it seems like a reasonable place to start.

I created and compiled the attached keymap with the online configurator, then flashed it to my xd75 with QMK Toolkit 0.0.12. Everything works as expected in Fedora 30 and macOS 10.12.6. On Windows 10 (1903/18362.329), however, the media keys don't work and layer 2 does not work. I tested media keys on a Logitech keyboard, and they work, which suggests Windows is configured correctly to respond to them.

[jpk-20190904.json.gz](https://github.com/qmk/qmk_configurator/files/3584450/jpk-20190904.json.gz)

yanfali, 2019-09-06:
You're using the macos specific keycodes. Try the more generic ones KC_VOLD and KC_VOLU

johann-koebbe, 2019-09-06:
Thanks @yanfali ! I'll try as soon as I reasonably can. :tada:

johann-koebbe, 2019-09-06:
@yanfali That worked great! Thanks so much. As for the layer not working, it turned out to be a PEBKAC. NumLock wasn't on. Derp. Thanks for great support!

closed 2019-09-06 by johann-koebbe