qmk/qmk_configurator #491: Generate an ascii art keymaps

Nanoseb, 2019-09-11:
We already have the option to export the keymaps as images. However it would be also nice to have an ascii art version of it to put in keymap.c, readme or else. 

Either something like this:
```
   ┌─────┬─────┬─────┬─────┬─────┬─────┬─────┐     ┌─────┬─────┬─────┬─────┬─────┬─────┬─────┬─────┐
   │ESC  │  1  │  2  │  3  │  4  │  5  │  6  │     │  7  │  8  │  9  │  0  │  -  │  =  │BSLS │ DEL │ 
   ├─────┴──┬──┴──┬──┴──┬──┴──┬──┴──┬──┴──┬──┘  ┌──┴──┬──┴──┬──┴──┬──┴──┬──┴──┬──┴──┬──┴──┬──┴─────┤
   │TAB     │  Q  │  W  │  E  │  R  │  T  │     │  Y  │  U  │  I  │  O  │  P  │  [  │  ]  │BSPC    │
   ├────────┴┬────┴┬────┴┬────┴┬────┴┬────┴┐    └┬────┴┬────┴┬────┴┬────┴┬────┴┬────┴┬────┴────────┤
   │CAPS     │  A  │  S  │  D  │  F  │  G  │     │  H  │  J  │  K  │  L  │  ;  │  '  │ENTER        │
   ├─────────┴──┬──┴──┬──┴──┬──┴──┬──┴──┬──┴──┐  └──┬──┴──┬──┴──┬──┴──┬──┴──┬──┴──┬──┴───────┬─────┤
   │LSHFT       │  Z  │  X  │  C  │  V  │  B  │     │  N  │  M  │  ,  │  .  │  /  │RSHFT     │ UP  │
   ├──────┬─────┴─┬───┴──┬──┴─────┴────┬┴─────┤     ├─────┴┬────┴────┬┴────┬┴────┬┴────┬─────┼─────┤
   │LCTRL │L_GUI  │L_ALT │SPC          │FN1   │     │SPC   │BSPC     │R_ALT│R_CTR│LEFT │DOWN │RIGHT│
   └──────┴───────┴──────┴─────────────┴──────┘     └──────┴─────────┴─────┴─────┴─────┴─────┴─────┘
```

Or probably easier to generate:
```
 +-------------------------------------------------------------------------+
 | ESC | 1 |  2 |  3 |  4 |  5 |  6 |  7 |  8 |  9 |  0 |  - |  = | BACKSP |
 +-------------------------------------------------------------------------+
 | TAB  |  Q |  W |  E |  R |  T |  Y |  U |  I |  O |  P |  [ |  ] |   \  |
 +-------------------------------------------------------------------------+
 | FN     |  A |  S |  D |  F |  G |  H |  J |  K |  L |  ; |  ' |  RETURN |
 +-------------------------------------------------------------------------+
 | LSHIFT   |  Z |  X |  C |  V |  B |  N |  M |  , |  . |  / | SHIFT | FN |
 +-------------------------------------------------------------------------+
 |LALT|LMETA|RCTRL|   SPACE | FN | SPACE           | RALT| MENU| RWIN|RCTRL|
 +-------------------------------------------------------------------------+
```

yanfali, 2019-09-15:
Could definitely try and do this, but it might be more useful to have it on the new QMK CLI since that is a python CLI tool.

Nanoseb, 2019-09-15:
Ok, interesting. I put the issue on this repository because to my (limited) knowledge qmk cli and other tools didn't know about the keyboard layout, when this configurator does. But if I am mistaken and this data is available to qmk cli, then I may even try to implement it myself (I can't do javascript, but am ok in python).

alexweininger, 2019-10-05:
This would be super nice when documenting keymap.c files! +1

sanxy, 2019-10-23:
Knowing the keyboard layout is essential, thanks

yanfali, 2019-10-25:
locking this issue until after hacktober