qmk/qmk_configurator #501: Clear Layer function breaks layout key count and key assignment if the function is run from Layer 0

noroadsleft, 2019-09-23:
## Steps to Reproduce

1. Load any keyboard in QMK Configurator
2. While Layer 0 is selected, click the Clear Layer button and confirm when the dialog prompt appears.

The key count that appears at the lower right of the keymap will read `0 Keys`.

Any of the following restore the key count:

- Select the last key of the layout while on Layer 0 and assign a keycode to it (assuming you know which key is the last one)
- Importing a keymap (manually, or switch to Keyboard Tester or Print Keymap and back)
- Change the currently selected layout (e.g. ANSI to ISO), if there is another option

yanfali, 2019-09-23:
Good find. This is an internal implementation detail leaking into the UI. I think the easiest way to fix it is use the layout macro count.

sandyjoshi, 2019-10-11:
@yanfali  is this issue fixed?

noroadsleft, 2019-10-11:
@sandyjoshi Yes, this issue was fixed in the pull request referenced above.

closed 2019-10-11 by noroadsleft