qmk/qmk_configurator #512: Dc-2014 Print Screen button

SneakyRobb, 2019-10-01:
Hi,

I am having an issue where my print screen button on the DC-2014 keyboard.

It seems to give the code 37 which is Pad_asterisk when I press print screen. 

Could you add this additional layout where that key is changed to the layout options for 

keyboard "converter_xt_usb?"

Thanks,
Robb

[converter_xt_usb_dc2014.txt](https://github.com/qmk/qmk_configurator/files/3677274/converter_xt_usb_dc2014.txt)

yanfali, 2019-10-01:
@SneakyRobb can't you just fix this in software? Print Screen is a standardized code. This seems like a keyboard specific issue.

yanfali, 2019-10-01:
@SneakyRobb you need to get these fixes into qmk/qmk_firmware configurator just builds what is upstream

closed 2019-10-01 by yanfali

noroadsleft, 2019-10-01:
@SneakyRobb 

> [converter_xt_usb_dc2014b.txt](https://github.com/qmk/qmk_configurator/files/3677317/converter_xt_usb_dc2014b.txt)

This file is assigning `KC_PAST` (Pad Asterisk) to the key that is labeled Print Screen on your keyboard.

Import this file to Configurator, click the X on the `KC_PAST` key, and assign Print Screen (`KC_PSCR`) in its place. Configurator is doing exactly what this file is telling it to do.

SneakyRobb, 2019-10-01:
Hi, 

I realise now this is a qmk issue and not an issue with the configurator. 

On the XT keyboard "print screen" and "asterisk" are on the same key and HID says the key is just  numpad asterisk.  

https://github.com/tmk/tmk_keyboard/issues/552
That is what was changed on the TMK xt_layout in this issue there. 

So If I use the qmk xt_usb layout to make a keymap, that key does not work. 

So to fix this, and use that key. I still used the qmk_configurator to make my layout. 
Then I downloaded the layout files, I downloaded and installed the qmk files, and copying the tmk code, I altered the relevant files locally. 

Then I used the msys2 environment with all the files to make the layout with that key as asterisk.

Here are the files I altered.

for the info.json - dont think necessary but did
[info.txt](https://github.com/qmk/qmk_configurator/files/3678214/info.txt)

For keymap.c in the default layout
[keymap.txt](https://github.com/qmk/qmk_configurator/files/3678221/keymap.txt)

and this for xt_usb.h
[xt_usb.txt](https://github.com/qmk/qmk_configurator/files/3678237/xt_usb.txt)

noroadsleft, 2019-10-02:
Going to move this to qmk/qmk_firmware; it's more appropriate there.