qmk/qmk_configurator #537: Not able to select alternative controllers for Atreus

RagingRoosevelt, 2019-10-10:
In the configurator, you're able to select Atreus variants (astar or teensy2) 
[image]
but it doesn't have any impact in the resulting compiled firmware.  In particular, the astar should result in a mirrored layout (using `PCBDOWN=yes` flag?) when selected.  At present, when compiling for astar, the layout is mirrored from what is displayed in the configurator.

---

In the firmware repo, there are already entries for the [astar keybaord variant](https://github.com/qmk/qmk_firmware/tree/master/keyboards/atreus/astar) and the [teensy2 keyboard variant](https://github.com/qmk/qmk_firmware/tree/master/keyboards/atreus/teensy2).  

In the public keymaps folder for the configurator, there's [an atreus default keymap](https://github.com/qmk/qmk_configurator/blob/master/public/keymaps/atreus_default.json).

I'm not sure how to solve this so that I could submit a pull request myself.  I'd be happy to do so with a bit of guidance.  I'm not sure if the right approach would be to add a new default layout with the line `"keyboard": "atreus/astar"` or if there's some other way to achieve this.

yanfali, 2019-10-10:
@skullydazed you want to weigh in on this?

skullydazed, 2019-10-24:
It looks like you have an earlier version of the astar pcb, where the PCB labels face down. Per the atreus readme:

> If your keyboard layout is a mirror image of what you expected (i.e. you do not get QWERTY on the left but YTREWQ on the right), then you have an A-Star powered Atreus (older than March 2016) with PCB labels facing down instead of up. Specify that by adding PCBDOWN=yes to your make commands

I think to solve this for configurator someone would need to create `atreus/astar/mirrored` which sets `PCBDOWN=yes`.

zvecr, 2019-10-24:
Fixed by #7137

closed 2019-10-24 by zvecr